cirosantilli/linux-kernel-module-cheat #5: ./run fails with "No rule to make target 'qemu_x86_64_defconfig'.  Stop."

cirosantilli, 2017-10-30:
Hi Ciro,
I Clone your repo from Github for the qemu , 

1.Uncomment all deb-src line from

    /etc/apt/sources.list

2.apt-get update

3. apt-get build-dep qemu 

4. git clone https://github.com/cirosantilli/linux-kernel-module-cheat.git

After that 

./run

is throwing error

```
root@virtual-machine:~/pci-driver/linux-kernel-module-cheat-master# ./run
make -C '/lib/modules/4.10.0-35-generic/build' M='/home/pci-driver/linux-kernel-module-cheat-master/kernel_module' clean
make[1]: Entering directory '/usr/src/linux-headers-4.10.0-35-generic'
make[1]: Leaving directory '/usr/src/linux-headers-4.10.0-35-generic'
make: Entering directory '/home/pci-driver/linux-kernel-module-cheat-master/kernel_module/user'
rm -f *'.out'
make: Leaving directory '/home/pci-driver/linux-kernel-module-cheat-master/kernel_module/user'
make: *** No rule to make target 'qemu_x86_64_defconfig'.  Stop.
```

cirosantilli, 2017-10-30:
Hi, did you update the submodules?

    git submodule update --init --recursive

or clone with:

    git clone --recursive

alokprasad, 2017-10-30:
HI Ciro,
There is some issue in my git , so actually downloaded master.zip https://github.com/cirosantilli/linux-kernel-module-cheat/archive/master.zip and then unzipped it, didnt use the git command at all.

cirosantilli, 2017-10-30:
The .zip does not contain submodules. To use it, you would have to go to every submodule and download it manually as well. I think buildroot has further submodules which makes it even harder. Get git working :-)

alokprasad, 2017-10-31:
Thanks , it worked and proceeded further , 
but i had make one change ...in 
kernel_module/make-host.sh
i had to manually put ncore value , while passing to it -"j"...

But now stucked at 
  GEN     /home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/Makefile
BR2_DEFCONFIG='' KCONFIG_AUTOCONFIG=/home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/buildroot-config/auto.conf KCONFIG_AUTOHEADER=/home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/buildroot-config/autoconf.h KCONFIG_TRISTATE=/home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/buildroot-config/tristate.config BR2_CONFIG=/home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/.config HOST_GCC_VERSION="5" BUILD_DIR=/home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build SKIP_LEGACY= /home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/buildroot-config/conf --silentoldconfig Config.in
>>> linux custom Syncing from source dir ../linux
ERROR: ../linux does not exist
package/pkg-generic.mk:163: recipe for target '/home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.stamp_rsynced' failed
make[1]: *** [/home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.stamp_rsynced] Error 1
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2

cirosantilli, 2017-10-31:
./linux/ is also a submodule, have you downloaded it as well?

Please ensure that you can run `git submodule update --init --recursive`.

alokprasad, 2017-10-31:
Yes currently doing it..looks like it huge..arodun 4gb..

cirosantilli, 2017-10-31:
Welcome to systems programming :-)

alokprasad, 2017-10-31:
Done ...Linux checkout ..
but now...throwing following error.
...
...
/bin/sed -i -e "/\\<CONFIG_KERNEL_XZ\\>/d" /home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.config
echo '# CONFIG_KERNEL_XZ is not set' >> /home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.config
/bin/sed -i -e "/\\<CONFIG_CPU_LITTLE_ENDIAN\\>/d" /home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.config
echo 'CONFIG_CPU_LITTLE_ENDIAN=y' >> /home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.config
# As the kernel gets compiled before root filesystems are
# built, we create a fake cpio file. It'll be
# replaced later by the real cpio archive, and the kernel will be
# rebuilt using the linux-rebuild-with-initramfs target.
/bin/sed -i -e "/\\<CONFIG_DEVTMPFS\\>/d" /home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.config
echo 'CONFIG_DEVTMPFS=y' >> /home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.config
/bin/sed -i -e "/\\<CONFIG_DEVTMPFS_MOUNT\\>/d" /home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.config
echo 'CONFIG_DEVTMPFS_MOUNT=y' >> /home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.config
make[2]: *** No rule to make target 'oldconfig'.  Stop.
linux/linux.mk:466: recipe for target '/home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.stamp_kconfig_fixup_done' failed
make[1]: *** [/home/embsys/pci-driver/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.stamp_kconfig_fixup_done] Error 2
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2

cirosantilli, 2017-10-31:
I have just done a full build from scratch with:

```
git clone --recursive https://github.com/cirosantilli/linux-kernel-module-cheat
cd linux-kernel-module-cheat
./run
```

Try to nuke your build and do the same.

Also provide your distro and version.

alokprasad, 2017-10-31:
Sure, wil ltry it again..

root@ubuntu:~# cat /etc/issue
Ubuntu 16.04.2 LTS 
root@ubuntu:~# uname -a
Linux ubuntu 4.8.0-58-generic #63~16.04.1-Ubuntu SMP Mon Jun 26 18:08:51 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

alokprasad, 2017-11-01:
Finally was able to succeed, But good enough amout of space is needed and one more package
apt-get install libsdl2-dev is required.

cirosantilli, 2017-11-01:
Glad it worked! Add the libsdl2-dev dependency to readme: https://github.com/cirosantilli/linux-kernel-module-cheat/commit/456bcce73f0c1d941ce5163c93bde66aa8eb30b5

closed 2017-11-01 by cirosantilli

cirosantilli, 2017-11-16:
BTW, I've not introduced shallow clones to save clone time: https://github.com/cirosantilli/linux-kernel-module-cheat/commit/23eddfb2a2242fa1bad2570a39779e4e0db497cc