cirosantilli/linux-kernel-module-cheat #6: test virtual device from guest OS

EvaChenn, 2017-11-01:
Thanks for your sharing  since there are seldom data of how to implement device in Qemu.
I have learned a lot from your post in this page [how-to-add-a-new-device-in-qemu-source-code](https://stackoverflow.com/questions/28315265/how-to-add-a-new-device-in-qemu-source-code) 
Recently, I have created a virtual device in qemu(add a c file under hw/gpio), and I have do "make" on this virtual device.
The next step is to test this virtual device by access the memory of this device form guest OS.
However, I stuck at this step.
I don't know how to call the device's read/write in Qemu from guest OS.
I just call readl(the add. used in sysbus_create_simple() ) , but it didn't work...
I think the problem I encounter may be the wrong memory access.
then I saw this page, you use
**iowrite32(0x9ABCDEF0, map + 4);**
to test device which is in Qemu.

May I ask why you use  0x9ABCDEF0 this address?
or do you have any suggestions for testing self-created virtual in Qemu?

Thanks anyway~~

cirosantilli, 2017-11-01:
Hi Eva,

I don't know much either, but here goes.

readX vs ioreadX: "explained" in LDD3: http://www.makelinux.net/ldd3/chp-9-sect-4 but I don't understand in great detail except that readX is bad and ioreadX is good.

 0x9ABCDEF0: just a random number that is easy to recognize for debugging. That thing is there at https://github.com/cirosantilli/linux-kernel-module-cheat/blob/26d0979aa602d212401f015765649ecd0442a516/kernel_module/platform_device.c#L40 just to silence the IRQ, which would keep going off until it gets a reply (TODO I don't know why it is like that yet in detail).

The corresponding platform device is a bit weird, I should improve / document the example better. At: https://github.com/cirosantilli/qemu/blob/e583d175e4cdfb12b4812a259e45c679743b32ad/hw/misc/lkmc_platform_device.c#L30 we see that any write to the memory region toggles the IRQ on / off.

Then the IRQ handler also write to that address to turn the IRQ off.

And the kernel module toggles it on at probe time: https://github.com/cirosantilli/linux-kernel-module-cheat/blob/26d0979aa602d212401f015765649ecd0442a516/kernel_module/platform_device.c#L88 to make the testing as simple as possible, just insert the module, and you can already observe the interrupt.

The actual address of the register is obtained at https://github.com/cirosantilli/linux-kernel-module-cheat/blob/26d0979aa602d212401f015765649ecd0442a516/kernel_module/platform_device.c#L71 whic h fetches the information from the device tree fie we put into the linux kernel at: https://github.com/cirosantilli/linux/blob/361bb623671a52a36a077a6dd45843389a687a33/arch/arm/boot/dts/versatile-pb.dts#L42 which matches the insertion point on QEMU: https://github.com/cirosantilli/qemu/blob/e583d175e4cdfb12b4812a259e45c679743b32ad/hw/arm/versatilepb.c#L311

cirosantilli, 2017-11-02:
Also, play around with GDB'ing QEMU. This is not hard to setup, but I also have a helper described at: https://github.com/cirosantilli/linux-kernel-module-cheat/blob/26d0979aa602d212401f015765649ecd0442a516/getting-started.md#debug-qemu

EvaChenn, 2017-11-06:
Sorry for late reply.
After studying your code for several days, I copy your files to my qemu & linux to understand the flow.
the step I conduct is as bellow:

### 1. Add qemu device driver
i.  add [qemu/hw/misc/lkmc_platform_device.c](https://github.com/EvaChenn/qemu-2.10/blob/master/hw/misc/lkmc_platform_device.c)
ii. add **obj-y += lkmc_platform_device.o** to [qemu/hw/misc/Makefile.objs
](https://github.com/EvaChenn/qemu-2.10/blob/master/hw/misc/Makefile.objs#L63)iii. add **CONFIG_LKMC_PLATFORM_DEVICE** to [qemu-2.10/default-configs/arm-softmmu.mak](https://github.com/EvaChenn/qemu-2.10/blob/master/default-configs/arm-softmmu.mak#L5)
iv. add **sysbus_create_simple("lkmc_platform_device", 0x101e9000, pic[18]);** to [qemu/hw/arm/versatilepb.c](https://github.com/EvaChenn/qemu-2.10/blob/master/hw/arm/versatilepb.c#L312)
v. make qemu

### 2. Add linux device driver
i.  add platform_device.c to [linux/driver/my_drvr/platform_device.c](https://github.com/EvaChenn/raspiKernel_for_qemu/blob/master/linux/drivers/my_drvr/platform_device.c)
ii. make driver as the steps methioned in [this article](https://stackoverflow.com/questions/11710022/adding-new-driver-code-to-linux-source-code)
iii. make kernel image for qemu by [build_kernel_qemu](https://github.com/EvaChenn/raspiKernel_for_qemu/blob/master/build-kernel-qemu)
// this build script is sepcialized for raspberrypi on Qemu with versatilepb platform
iv. add dts to linux/arch/arm/boot/dts/versatile-pb.dts

### 3. open raspberrypi on Qemu with versatilepb
use [test.sh](https://github.com/EvaChenn/raspiKernel_for_qemu/blob/master/test.sh) to start Qemu

I successfully open the raspkernel on Qemu, and have seen the string "lkmc_platform_device_init" by dmesg
but I don't konw when the lkmc_platform_device_probe() function will be called, or how to invoke this function so that i can test to do some operation to the QEMU lkmc_platform_device.

Do you have any advice? or is there any step that I miss or conduct in wrong way?
Thanks for your help sincerely.

cirosantilli, 2017-11-06:
I think `lkmc_platform_device_probe` is called by the kernel once the DTS is well installed.

The DTB is given on the QEMU command line with -dtb , it looks like your script is not passing it.

BTW, there is now the `-M raspi` option if you are interested in emulating the raspberry pi: https://stackoverflow.com/questions/38837606/emulate-raspberry-pi-raspbian-with-qemu/45893057#45893057 but I haven't been able to get display and network working yet.

EvaChenn, 2017-11-06:
I conduct the command to create  versatile-pb.dts
**dtc -O dtb -I dts -o versatilepb.dtb ../linux/arch/arm/boot/dts/versatile-pb.dts**

then add **-dtb versatilepb.dtb** to qemu command line [test.sh](https://github.com/EvaChenn/raspiKernel_for_qemu/blob/master/test.sh)

however, I got this error "Neither atags nor dtb found" and kernel panic in QEMU
 
[image]

I am wondering this error is caused by the special kernel for raspberrypi on versatilepb...
do you have any idea of this? or what document should I dig more?

------------
I have tried **-M raspi** already and the reason I don't use it is the lack of network.
However, I must emulate raspi which is a IOT-related board for research about IOT, and the network is the necessary component.
network works in versatilepb and there is a build script to build a special raspberrypi kernel for versatilepb platform
Therefore, I chose to use **-M versatilepb** to simulate raspberrypi.

Thanks for your reply.

cirosantilli, 2017-11-06:
OK, I'm a bit clueless now :-)

I would try to generate the dtb using the linux kernel makefile which is what Buildroot is doing for me.

I would also disassemble the dtb to see if it looks OK: https://stackoverflow.com/questions/14000736/tool-to-visualize-the-device-tree-file-dtb-used-by-the-linux-kernel/39931834#39931834

EvaChenn, 2017-11-09:
[versatile-pb_dis.dt](https://github.com/EvaChenn/raspiKernel_for_qemu/blob/master/versatile-pb_dis.dts)
this is my device tree that I use **dtc -I dtb -O dts -o versatile-pb_dis.dts versatilepb.dtb**
(I already take off the part of lkmc device in versatilepb.dtb)
I am wondering if my dts file looks the same as yours?
cause my linux can not use **make dtbs** to create dtb
so I use dtc command to create dtb, and I am wondering if my dtb is wrong

sorry for keep bothering you, thanks for all your help!!

EvaChenn, 2017-11-11:
now I am trying to add a linux driver on build root, which the kernel I used is normal kernel instead of kernel for raspberry, to test if I establish virtual device on qemu right or not 
may I ask how to add module to linux which is in the buildroot?
I have read [this page](https://stackoverflow.com/questions/40307328/how-to-add-a-linux-kernel-driver-module-as-a-buildroot-package) 
but I don't understand how this module connect to the qemu device without adding the information in dtb :'(

cirosantilli, 2017-11-11:
I don't understand what part you don't understand :-)

https://stackoverflow.com/questions/40307328/how-to-add-a-linux-kernel-driver-module-as-a-buildroot-package describes how to compile the .ko It that works, you should see the .ko on target filesystem, but then you still have to insmod it after boot.

The device finds its DTB entry using the identifier `"lkmc_platform_device"` at:

```
static const struct of_device_id of_lkmc_platform_device_match[] = {
	{ .compatible = "lkmc_platform_device", },
	{},
};
```

and then it queries the DTB found with functions of form `of-*` such as `of_property_read_u32(np, "lkmc-asdf", &asdf)`.

EvaChenn, 2017-11-13:
Sorry for not having described my question clearly.
I have already seen the .ko on target filesystem and insmod it after boot.
But how can I test if I have linked the dirver on linux to driver on qemu correctly?
After insmod, I have see the "lkmc_platform_device_init" string, but it doesn't mean I linked the driver between linux and qemu well.
Apart from this string, I used "dmesg" see noting related to lkmc, which meant that the module in linux didn't do lkmc_platform_device_probe() function.
Am I doing right?

thank you so much!!

cirosantilli, 2017-11-13:
Try the two following things:

- on guest, examine `/proc/device-tree` to ensure that the lkmc device is actually there: https://unix.stackexchange.com/questions/265890/is-it-possible-to-get-the-information-for-a-device-tree-using-sys-of-a-running/330926#330926

- on host, run QEMU through GDB, and put a breakpoint on the lkmc device init functio to ensure that you actually compiled with it.

EvaChenn, 2017-11-14:
Thanks for those two methods, they really help me a lot.
 I finally link the driver on qemu to the module on linux!!

But I still stuck at using dtb in qemu-raspberrypi-kernel now...
I have googled this problem for a long time and have seen that you also dedicated to the subject about emulating raspberrypi on qemu.
Have you used [this kernel](https://github.com/dhruvvyas90/qemu-rpi-kernel) with dtb successfully?

cirosantilli, 2017-11-14:
I did run dhruvvyas90/qemu-rpi-kernel on `-M versatilepb` as described at:  https://raspberrypi.stackexchange.com/questions/165/emulation-on-a-linux-pc/53991#53991 but I never tried to add QEMU virtual devices to it. I'd expect it to be the exact same process as done in this repo.

EvaChenn, 2017-11-20:
Sorry for late reply.
I finally successfully create a driver on qemu-raspberrypi-kernel and match it to a qemu virtual device!!!!
Since that qemu-raspberrypi-kernel use "platform device" structure to describe hardware setting instead of the device tree, I add lkmc device by platform_device_register() then it works!!!
Thanks for all your help. :)

cirosantilli, 2017-11-20:
Great news! Thanks for the `platform_device_register` info, I will keep that in mind for the future.

closed 2017-11-20 by cirosantilli