cirosantilli/linux-kernel-module-cheat #7: Error when configuring submodules - "error: no such remote ref"

tompreston, 2017-12-04:
Hi, this looks really useful but I'm having an issue on Debian stable when running `./configure`:

```
tpreston:~/linux-kernel-module-cheat$ ./configure 
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: no such remote ref 8ce27bb9fee80a406a4199657ef90e3c315e7457
Fetched in submodule path 'buildroot', but it did not contain 8ce27bb9fee80a406a4199657ef90e3c315e7457. Direct fetching of that commit failed.
```

That ref _does_ exist, and I can clone it manually:
```
tpreston:~/linux-kernel-module-cheat$ git clone git://git.buildroot.net/buildroot buildroot
```

But then the same error appears with another submodule:
```
tpreston:~/linux-kernel-module-cheat$ ./configure 
Submodule path 'buildroot': checked out '8ce27bb9fee80a406a4199657ef90e3c315e7457'
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: no such remote ref d4160b40c586e36c77600bf2a71ee00fc0a3e8a7
Fetched in submodule path 'linux', but it did not contain d4160b40c586e36c77600bf2a71ee00fc0a3e8a7. Direct fetching of that commit failed.
```

Not sure if there's an issue with the shallow clone of submodules. I'm not quite sure what's going on there.

Removing `--depth` doesn't work either:
```
diff --git a/configure b/configure
index 7407827..f561c25 100755
--- a/configure
+++ b/configure
@@ -18,7 +18,7 @@ shift $(($OPTIND - 1))
 # In particular:
 # - `shallow = true` on the submodule has no effect for the non default educational branches of our submodules
 # - QEMU's submodules point to commits that are neither under branches nor tags, and so `--shallow-submodules` fails
-git submodule update --depth 1 --jobs 4 --init
+git submodule update --jobs 4 --init
 cd qemu
 git submodule update --init
 ```

Any advice? I'd really like to use this.

cirosantilli, 2017-12-04:
Hi Tom, can you try to do a new non shallow clone with:

```
git clone --recursive https://github.com/cirosantilli/linux-kernel-module-cheat
```

?

The shallow clone can save considerable time / disk space, but as the comments in `./configure` indicate, the feature is flaky as hell...

What is your Git version? Is it 2.11 https://packages.debian.org/stretch/git ?

I tested in 2.14 Ubuntu 16.04 just now and it worked. If you have the patience, try to compile Git from source / get a newer version somehow so we can decide if it is a fixed Git bug or something else.

mcondarelli, 2018-01-08:
I have the same problem with "git version 2.11.0" (debian stable standard).
Doing a full recursive clone cured the problem (but with LOOOONNNNGGGGG times).
I hope You can fix module handling...

cirosantilli, 2018-01-08:
@mcondarelli  thanks for the confirmation.

This was the best setup I could find unfortunately, I don't know how to improve this further except having an automatic fallback mechanism to full recursive initialization (which as you noticed, takes a long time :-()

Let me know if you find something better.

cirosantilli, 2018-01-31:
I'll be closing this with https://github.com/cirosantilli/linux-kernel-module-cheat/commit/cf063ad8d20d18c37fbfb148ae8120d32adcfa3c for now, let me know if you see a better solution.

closed 2018-01-31 by cirosantilli