cirosantilli/linux-kernel-module-cheat #10: WARNING: 'aclocal-1.15' is missing on your system.

poweihuang17, 2018-01-20:
Hi, I tried to build it on another machine, and this time, when I'm building hostflex, it showed this message.
After I followed this:
https://stackoverflow.com/questions/33278928/how-to-overcome-aclocal-1-15-is-missing-on-your-system-warning-when-compilin

I install autoconf, and this error is resolved.
Do you need to add autoconf as dependency?

cirosantilli, 2018-01-20:
What is the OS on both systems? Which version?

On Ubuntu 17.10, `aclocal-1.15` is part of the `automake` package. This can be found with `apt-file search aclocal-1.15`, so the solution on this system would be `sudo apt-get install automake`.

I added it to the `./configure` as well, I thought it was already installed by `build-essential` but it doesn't seem to be the case according to `apt-rdepends`: https://askubuntu.com/questions/468753/recursive-dependencies

Fixed at: https://github.com/cirosantilli/linux-kernel-module-cheat/commit/1be85b6dc2a16d6de2138c264c6c169359c3c6e4

poweihuang17, 2018-01-20:
My first server is Ubuntu 16.04.3 LTS (GNU/Linux 4.10.0-38-generic x86_64).
My second one is Ubuntu 16.04.3 LTS (GNU/Linux 4.10.0-28-generic x86_64)

poweihuang17, 2018-01-20:
Yes, I used the same one "sudo apt-get install automake" to solve it.

poweihuang17, 2018-01-20:
So, while I didn't test your commit from scratch, it seems to fix the issue.  Thanks.

closed 2018-01-20 by poweihuang17