cirosantilli/linux-kernel-module-cheat #9: mount: mounting debugfs on /sys/kernel/debug failed: No such file or directory

poweihuang17, 2018-01-20:
Hi, when I followed the guide and open up qemu, it has this error message and I couldn't use ftrace.
Do you know how to debug this problem?

cirosantilli, 2018-01-20:
`mkdir /sys/kernel/debug` and then `mount -t debugfs none /sys/kernel/debug` should solve it.

debugfs is a filesystem and is mounted like any other. That is the same error message we get when trying to mount on desktop without an existing dir.

The question however is why the `fstab` entry didn't work for you. It worked for me automatically.

poweihuang17, 2018-01-20:
Thanks, but when I wanted to use mkdir /sys/kernel/debug, it shows "mkdir: can't create directory '/sys/kernel/debug': Operation not permitted"...

----
adding dns 10.0.2.3
OK
S98

buildroot login: root                   
# ls
# mkdir /sys/kernel/debug
mkdir: can't create directory '/sys/kernel/debug': Operation not permitted

cirosantilli, 2018-01-20:
Hmm. As a workaround, create debugfs somewhere else, e.g. `mkdir /debugfs`. But I'd like to understand things better at some poing.

poweihuang17, 2018-01-20:
Ok. It shows "mount: mounting none on /debugfs failed: No such device" now... 
However, thanks for your help.

cirosantilli, 2018-01-20:
Fixed by: f5a0111c3e0edec499b8e0e3c1c6efe89e7ceed8

It was ignoring the kernel configuration because I removed the quotes for some reason, so debugfs was never enabled.

When debugfs is enabled, the kernel creates `/sys/kernel/debugfs` by default (but does not mount it).

Thank you very much for the report!

cirosantilli, 2018-01-20:
To try this out, do a clean build with:

`rm -rf  buildroot/output.x86_64~`

closed 2018-01-20 by cirosantilli

poweihuang17, 2018-01-20:
Thanks. Right now when I compile and run it, it shows this under /sys/kernel/debug. So, it seems to be fixed. Thanks for your effort. This is a really good repo!.
# ls
acpi                gpio                sleep_time
bdi                 hid                 suspend_stats
block               kernel_page_tables  tracing
clk                 mce                 usb
dma_buf             pm_qos              virtio-ports
dri                 regmap              wakeup_sources
extfrag             sched_debug         x86

Jialufeng, 2019-09-27:
[image]
???

cirosantilli, 2019-09-27:
@Jialufeng hi, is this in repo? I didn't know about the `debugfs` command until today, and it has nothing to do with `/sys/kernel/debug` apparently. Try asking on unix stack exchange.

lix19937, 2024-10-16:
```
root@tegra-ubuntu:/sys/kernel# sudo mkdir debug
mkdir: cannot create directory ‘debug’: Operation not permitted

root@tegra-ubuntu:/sys/kernel# mount |grep sysfs
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime,seclabel)
root@tegra-ubuntu:/sys/kernel# zcat /proc/config.gz |grep CONFIG_DEBUG_FS
# CONFIG_DEBUG_FS is not set

```