cirosantilli/linux-kernel-module-cheat #11: ./configure says "usage: git submodule"

bootrino, 2018-01-31:
On Ubuntu 16

When I put in ./configure it says:

```
root@ubuntu-s-1vcpu-1gb-nyc1-01:~/tmp/linux-kernel-module-cheat# ./configure
usage: git submodule [--quiet] add [-b <branch>] [-f|--force] [--name <name>] [--reference <repository>] [--] <repository> [<path>]
   or: git submodule [--quiet] status [--cached] [--recursive] [--] [<path>...]
   or: git submodule [--quiet] init [--] [<path>...]
   or: git submodule [--quiet] deinit [-f|--force] [--] <path>...
   or: git submodule [--quiet] update [--init] [--remote] [-N|--no-fetch] [-f|--force] [--checkout|--merge|--rebase] [--reference <repository>] [--recursive] [--] [<path>...]
   or: git submodule [--quiet] summary [--cached|--files] [--summary-limit <n>] [commit] [--] [<path>...]
   or: git submodule [--quiet] foreach [--recursive] <command>
   or: git submodule [--quiet] sync [--recursive] [--] [<path>...]
root@ubuntu-s-1vcpu-1gb-nyc1-01:~/tmp/linux-kernel-module-cheat#
```

cirosantilli, 2018-01-31:
What is your git version?

Try to do a `git submodule update --init` manually.

Then comment out the `git submodule` commands on the `./configure` and run it again.

It seems that your version of git does not have `--depth 1` option (which I use to speed clone up).

bootrino, 2018-01-31:
Strange it does not seem to have a problem after doing what you suggest.

git version 2.7.4 on Ubuntu 16, as seen below

```
root@ubuntu-s-1vcpu-1gb-nyc1-01:~/linux-kernel-module-cheat# git --version
git version 2.7.4
root@ubuntu-s-1vcpu-1gb-nyc1-01:~/linux-kernel-module-cheat# git submodule update --init
Submodule 'buildroot' (https://github.com/buildroot/buildroot) registered for path 'buildroot'
Submodule 'linux' (https://github.com/cirosantilli/linux) registered for path 'linux'

```

cirosantilli, 2018-01-31:
I'll be closing this with https://github.com/cirosantilli/linux-kernel-module-cheat/commit/cf063ad8d20d18c37fbfb148ae8120d32adcfa3c for now, let me know if you see a better solution.

closed 2018-01-31 by cirosantilli