cirosantilli/linux-kernel-module-cheat #13: ./build gives an error

bootrino, 2018-02-01:
root@ubuntu-s-1vcpu-1gb-nyc1-01:~/linux-kernel-module-cheat# ./build
make: invalid option -- '1'
Usage: make [options] [target] ...
Options:
  -b, -m                      Ignored for compatibility.
  -B, --always-make           Unconditionally make all targets.
  -C DIRECTORY, --directory=DIRECTORY
                              Change to DIRECTORY before doing anything.
  -d                          Print lots of debugging information.
  --debug[=FLAGS]             Print various types of debugging information.
  -e, --environment-overrides
                              Environment variables override makefiles.
  --eval=STRING               Evaluate STRING as a makefile statement.
  -f FILE, --file=FILE, --makefile=FILE
                              Read FILE as a makefile.
  -h, --help                  Print this message and exit.
  -i, --ignore-errors         Ignore errors from recipes.
  -I DIRECTORY, --include-dir=DIRECTORY
                              Search DIRECTORY for included makefiles.
  -j [N], --jobs[=N]          Allow N jobs at once; infinite jobs with no arg.
  -k, --keep-going            Keep going when some targets can't be made.
  -l [N], --load-average[=N], --max-load[=N]
                              Don't start multiple jobs unless load is below N.
  -L, --check-symlink-times   Use the latest mtime between symlinks and target.
  -n, --just-print, --dry-run, --recon
                              Don't actually run any recipe; just print them.
  -o FILE, --old-file=FILE, --assume-old=FILE
                              Consider FILE to be very old and don't remake it.
  -O[TYPE], --output-sync[=TYPE]
                              Synchronize output of parallel jobs by TYPE.
  -p, --print-data-base       Print make's internal database.
  -q, --question              Run no recipe; exit status says if up to date.
  -r, --no-builtin-rules      Disable the built-in implicit rules.
  -R, --no-builtin-variables  Disable the built-in variable settings.
  -s, --silent, --quiet       Don't echo recipes.
  -S, --no-keep-going, --stop
                              Turns off -k.
  -t, --touch                 Touch targets instead of remaking them.
  --trace                     Print tracing information.
  -v, --version               Print the version number of make and exit.
  -w, --print-directory       Print the current directory.
  --no-print-directory        Turn off -w, even if it was turned on implicitly.
  -W FILE, --what-if=FILE, --new-file=FILE, --assume-new=FILE
                              Consider FILE to be infinitely new.
  --warn-undefined-variables  Warn when an undefined variable is referenced.

This program built for x86_64-pc-linux-gnu
Report bugs to <bug-make@gnu.org>
make: invalid option -- '1'
Usage: make [options] [target] ...
Options:
  -b, -m                      Ignored for compatibility.
  -B, --always-make           Unconditionally make all targets.
  -C DIRECTORY, --directory=DIRECTORY
                              Change to DIRECTORY before doing anything.
  -d                          Print lots of debugging information.
  --debug[=FLAGS]             Print various types of debugging information.
  -e, --environment-overrides
                              Environment variables override makefiles.
  --eval=STRING               Evaluate STRING as a makefile statement.
  -f FILE, --file=FILE, --makefile=FILE
                              Read FILE as a makefile.
  -h, --help                  Print this message and exit.
  -i, --ignore-errors         Ignore errors from recipes.
  -I DIRECTORY, --include-dir=DIRECTORY
                              Search DIRECTORY for included makefiles.
  -j [N], --jobs[=N]          Allow N jobs at once; infinite jobs with no arg.
  -k, --keep-going            Keep going when some targets can't be made.
  -l [N], --load-average[=N], --max-load[=N]
                              Don't start multiple jobs unless load is below N.
  -L, --check-symlink-times   Use the latest mtime between symlinks and target.
  -n, --just-print, --dry-run, --recon
                              Don't actually run any recipe; just print them.
  -o FILE, --old-file=FILE, --assume-old=FILE
                              Consider FILE to be very old and don't remake it.
  -O[TYPE], --output-sync[=TYPE]
                              Synchronize output of parallel jobs by TYPE.
  -p, --print-data-base       Print make's internal database.
  -q, --question              Run no recipe; exit status says if up to date.
  -r, --no-builtin-rules      Disable the built-in implicit rules.
  -R, --no-builtin-variables  Disable the built-in variable settings.
  -s, --silent, --quiet       Don't echo recipes.
  -S, --no-keep-going, --stop
                              Turns off -k.
  -t, --touch                 Touch targets instead of remaking them.
  --trace                     Print tracing information.
  -v, --version               Print the version number of make and exit.
  -w, --print-directory       Print the current directory.
  --no-print-directory        Turn off -w, even if it was turned on implicitly.
  -W FILE, --what-if=FILE, --new-file=FILE, --assume-new=FILE
                              Consider FILE to be infinitely new.
  --warn-undefined-variables  Warn when an undefined variable is referenced.

This program built for x86_64-pc-linux-gnu
Report bugs to <bug-make@gnu.org>
root@ubuntu-s-1vcpu-1gb-nyc1-01:~/linux-kernel-module-cheat#

cirosantilli, 2018-02-01:
Hmm, I can't reproduce this. Can you track down which `make` is failing? E.g. with `bash -x`. Something appears to be calling `make -1`.

bootrino, 2018-02-01:
I start with a clean ubuntu instance booted on Digital Ocean.

Here are all the commands that lead to the above error:

```
root@ubuntu-s-1vcpu-2gb-nyc1-01:~/linux-kernel-module-cheat# history
    1  ls
    2  apt update
    3  git clone https://github.com/cirosantilli/linux-kernel-module-cheat
    4  cd linux-kernel-module-cheat/
    5  ./configure -t
    6  sudo apt-get install gcc-6 g++-6
    7  sudo add-apt-repository ppa:ubuntu-toolchain-r/test
    8  apt update
    9  apt-get install gcc-6 g++-6
   10  ./configure
   11  ./build
   12  history
root@ubuntu-s-1vcpu-2gb-nyc1-01:~/linux-kernel-module-cheat#
```

And here is the full output of the above commands:

```
Welcome to Ubuntu 16.04.3 LTS (GNU/Linux 4.4.0-112-generic x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

  Get cloud support with Ubuntu Advantage Cloud Guest:
    http://www.ubuntu.com/business/services/cloud

0 packages can be updated.
0 updates are security updates.

The programs included with the Ubuntu system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Ubuntu comes with ABSOLUTELY NO WARRANTY, to the extent permitted by
applicable law.

Changing password for root.
(current) UNIX password:
Enter new UNIX password:
Retype new UNIX password:
root@ubuntu-s-1vcpu-2gb-nyc1-01:~# ls
root@ubuntu-s-1vcpu-2gb-nyc1-01:~# apt update
Get:1 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]
Hit:2 http://nyc2.mirrors.digitalocean.com/ubuntu xenial InRelease
Get:3 http://nyc2.mirrors.digitalocean.com/ubuntu xenial-updates InRelease [102 kB]
Get:4 http://nyc2.mirrors.digitalocean.com/ubuntu xenial-backports InRelease [102 kB]
Get:5 http://nyc2.mirrors.digitalocean.com/ubuntu xenial/main Sources [868 kB]
Get:6 http://nyc2.mirrors.digitalocean.com/ubuntu xenial/restricted Sources [4,808 B]
Get:7 http://nyc2.mirrors.digitalocean.com/ubuntu xenial/universe Sources [7,728 kB]
Get:8 http://nyc2.mirrors.digitalocean.com/ubuntu xenial/multiverse Sources [179 kB]
Get:9 http://nyc2.mirrors.digitalocean.com/ubuntu xenial/universe amd64 Packages [7,532 kB]
Get:10 http://nyc2.mirrors.digitalocean.com/ubuntu xenial/universe Translation-en [4,354 kB]
Get:17 http://security.ubuntu.com/ubuntu xenial-security/main Sources [107 kB]
Get:11 http://nyc2.mirrors.digitalocean.com/ubuntu xenial/multiverse amd64 Packages [144 kB]
Get:19 http://security.ubuntu.com/ubuntu xenial-security/restricted Sources [2,116 B]
Get:12 http://nyc2.mirrors.digitalocean.com/ubuntu xenial/multiverse Translation-en [106 kB]
Get:13 http://nyc2.mirrors.digitalocean.com/ubuntu xenial-updates/main Sources [291 kB]
Get:14 http://nyc2.mirrors.digitalocean.com/ubuntu xenial-updates/restricted Sources [2,524 B]
Get:15 http://nyc2.mirrors.digitalocean.com/ubuntu xenial-updates/universe Sources [189 kB]
Get:16 http://nyc2.mirrors.digitalocean.com/ubuntu xenial-updates/multiverse Sources [7,972 B]
Get:18 http://nyc2.mirrors.digitalocean.com/ubuntu xenial-updates/main amd64 Packages [710 kB]
Get:20 http://security.ubuntu.com/ubuntu xenial-security/universe Sources [49.3 kB]
Get:21 http://nyc2.mirrors.digitalocean.com/ubuntu xenial-updates/main Translation-en [295 kB]
[... 2021 lines cut ...]
  -v, --version               Print the version number of make and exit.
  -w, --print-directory       Print the current directory.
  --no-print-directory        Turn off -w, even if it was turned on implicitly.
  -W FILE, --what-if=FILE, --new-file=FILE, --assume-new=FILE
                              Consider FILE to be infinitely new.
  --warn-undefined-variables  Warn when an undefined variable is referenced.

This program built for x86_64-pc-linux-gnu
Report bugs to <bug-make@gnu.org>
root@ubuntu-s-1vcpu-2gb-nyc1-01:~/linux-kernel-module-cheat# history
    1  ls
    2  apt update
    3  git clone https://github.com/cirosantilli/linux-kernel-module-cheat
    4  cd linux-kernel-module-cheat/
    5  ./configure -t
    6  sudo apt-get install gcc-6 g++-6
    7  sudo add-apt-repository ppa:ubuntu-toolchain-r/test
    8  apt update
    9  apt-get install gcc-6 g++-6
   10  ./configure
   11  ./build
   12  history
root@ubuntu-s-1vcpu-2gb-nyc1-01:~/linux-kernel-module-cheat#

```

dakami, 2018-02-02:
Maybe change the first line of ./build from /usr/bin/env bash to:

#!/bin/bash -x

and send that output?