cirosantilli/linux-kernel-module-cheat #16: Error building kernel module "timer"

poweihuang17, 2018-02-08:
Hi,
When I made "git pull" today, the error shows:
/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.c:23:31: error: macro "DEFINE_TIMER" requires 4 arguments, but only 2 given
 DEFINE_TIMER(mytimer, callback);
                               ^
/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.c:23:1: error: data definition has no type or storage class [-Werror]
 DEFINE_TIMER(mytimer, callback);
 ^~~~~~~~~~~~
/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.c:23:1: error: type defaults to ‘int’ in declaration of ‘DEFINE_TIMER’ [-Werror=implicit-int]
/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.c: In function ‘callback’:
/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.c:28:13: error: mytimer’ undeclared (first use in this function)
  mod_timer(&mytimer, jiffies + onesec);
             ^~~~~~~
/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.c:28:13: note: each undeclared identifier is reported only once for each function it appears in
/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.c: In function ‘myinit’:
/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.c:34:13: error: mytimer’ undeclared (first use in this function)
  mod_timer(&mytimer, jiffies + onesec);
             ^~~~~~~
/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.c: In function ‘myexit’:
/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.c:40:13: error: mytimer’ undeclared (first use in this function)
  del_timer(&mytimer);
             ^~~~~~~
At top level:
/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.c:25:13: error: callback’ defined but not used [-Werror=unused-function]
 static void callback(struct timer_list *data)
             ^~~~~~~~
cc1: all warnings being treated as errors
scripts/Makefile.build:320: recipe for target '/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.o' failed
make[3]: *** [/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/./timer.o] Error 1
Makefile:1503: recipe for target '_module_/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/.' failed
make[2]: *** [_module_/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/.] Error 2
package/pkg-generic.mk:227: recipe for target '/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/.stamp_built' failed
make[1]: *** [/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/kernel_module-1.0/.stamp_built] Error 2
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2

cirosantilli, 2018-02-08:
Try `git submodule update` that interface changed when I updated the kernel to 4.15 recently.

poweihuang17, 2018-02-08:
Now, it has this error...
Makefile:932: *** "Cannot generate ORC metadata for CONFIG_UNWINDER_ORC=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel".  Stop.
package/pkg-generic.mk:227: recipe for target '/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.stamp_built' failed
make[1]: *** [/media/newslab/newslab/powei/riscv/learn_linux_kernel/cheat/linux-kernel-module-cheat/buildroot/output.x86_64~/build/linux-custom/.stamp_built] Error 2
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2

cirosantilli, 2018-02-08:
`sudo apt-get install libelf-dev` as indicated by the error message :-)

I also added this to `configure` during the 4.15 migration but forgot to tell you to rerun `configure`: https://github.com/cirosantilli/linux-kernel-module-cheat/blob/dc495f9334e38bfb864305d19f3b0f55f7a15d35/configure#L53

poweihuang17, 2018-02-08:
Thanks. I could successfully build it now.

closed 2018-02-08 by poweihuang17