cirosantilli/linux-kernel-module-cheat #17: Questions: How to send signal to process inside QEMU?

poweihuang17, 2018-02-08:
Hi,
Thanks for your always help. Do you know how to send signal to process inside QEMU?
I'm using non-graphic mode like this:
./run -n
 ./buildroot/output.x86_64~/host/usr/bin/qemu-system-x86_64 -m 128M -monitor telnet::45454,server,nowait -netdev user,hostfwd=tcp::45455-:45455,id=net0 -smp 1  -M pc -append 'root=/dev/vda nopat nokaslr norandmaps printk.devkmsg=on printk.time=y console=ttyS0' -device edu -device lkmc_pci_min -device virtio-net-pci,netdev=net0 -drive file=./buildroot/output.x86_64~/images/rootfs.ext2,if=virtio,format=raw -kernel ./buildroot/output.x86_64~/images/bzImage  -nographic
---
I found this link, but do you know which should I connect to for "telnet::45454"?

poweihuang17, 2018-02-08:
BTW, I'm using no-graphic mode.

cirosantilli, 2018-02-08:
Do you mean like hitting `Ctrl + C` to send SIGKILL to the foreground process in nographic mode?

Unfortunately I don't know how to do that well, it just kills QEMU itself, this is mentioned at the end of this section: https://github.com/cirosantilli/linux-kernel-module-cheat/blob/c771b40913eddd22809dc5504e44a1f3da9a4ede/getting-started.md#text-mode which link to Stack Exchange questions.

The best I can do for now is put on background with `sleep 10 $` and then `kill $!` or the like.

`telnet::45454` is for the QEMU monitor, it allows interaction with QEMU itself, e.g. `echo quit | ./qemumonitor` to quit QEMU.

But the monitor has no knowledge of Linux or processes, so you shouldn't be able to send signals from there.

See this for a list of monitor commands: https://en.wikibooks.org/wiki/QEMU/Monitor