cirosantilli/linux-kernel-module-cheat #18: Submodule don't initialize properly

scryver, 2018-02-11:
At the line:

https://github.com/cirosantilli/linux-kernel-module-cheat/blob/ae9bbf9aa641706dc618a961839fd9f4705d2281/configure#L23

Remove the 1 after --depth to make it work for newly cloned repos.

cirosantilli, 2018-02-11:
See also: https://github.com/cirosantilli/linux-kernel-module-cheat/issues/7 and on `configure` comments`.

I to make a fallback to full clone for git versions that don't support it properly at: https://github.com/cirosantilli/linux-kernel-module-cheat/blob/ae9bbf9aa641706dc618a961839fd9f4705d2281/configure#L27 but I can't test it properly since I don't have the older gits.

What happens to you exactly? Does the fallback get run, or does something fail? What is the last line that gets run, and what is the stdout?

scryver, 2018-02-11:
It looks exactly like the issues you mentioned. The git submodule only get the latest ref (of the master branch) and don't pull in the commit coupled with this repo. With the result that the commit that wants to be pulled in doesn't exist in the local git history.

Stdout:

```
Submodule 'buildroot' (https://github.com/buildroot/buildroot) registered for path 'buildroot'
Submodule 'linux' (https://github.com/cirosantilli/linux) registered for path 'linux'
Submodule 'qemu' (https://github.com/cirosantilli/qemu) registered for path 'qemu'
Cloning into '/code/linux-kernel-module-cheat/buildroot'...
Cloning into '/code/linux-kernel-module-cheat/linux'...
Cloning into '/code/linux-kernel-module-cheat/qemu'...
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: no such remote ref 8ce27bb9fee80a406a4199657ef90e3c315e7457
Fetched in submodule path 'buildroot', but it did not contain 8ce27bb9fee80a406a4199657ef90e3c315e7457. Direct fetching of that commit failed.
error: no such remote ref 8ce27bb9fee80a406a4199657ef90e3c315e7457
Fetched in submodule path 'buildroot', but it did not contain 8ce27bb9fee80a406a4199657ef90e3c315e7457. Direct fetching of that commit failed.
```
If I run ``git log --oneline`` in the directory buildroot I get:
```
7c384c3 qt5tools: fix typo in <pkg>_SOURCE
```
And that's all the history it knows, so it pulls in 1 depth as specified but it is the origin/master ref not the one referenced by the submodule. Hope this helps with finding a solution.