cirosantilli/linux-kernel-module-cheat #22: How to add my own binaries into buildroot?

lvcargnini, 2018-04-06:
Hi,

Please, how do I add my own binaries into the image created by buildroot?
Or
How do I add my code to be compiled and inserted into the create image?
I have my own set of benchmarks I would like to add into the image to test in x86_64 and Aarch64 later on.

cirosantilli, 2018-04-08:
Hi Vitorio, sorry for late reply!

Have a look at these pointers:

- https://github.com/cirosantilli/linux-kernel-module-cheat/tree/105c506ac639e89318621f7df89609f0c9525f26#add-new-buildroot-packages
- https://github.com/cirosantilli/linux-kernel-module-cheat/commit/ecaccfd8d0e550b3ec79dd3a2bc9363249e304db

You can then just copy your binaries into TARGET_DIR instead of compiling them.

But don't forget that they must have been compiled with `-static`.

Let me know if something is not clear.

cirosantilli, 2018-04-09:
> too much work for irq4

I've seen this, I don't kon't know how to solve it

> sh: memtest: not found

You haven't compiled with `-static`. The message is misleading, what it can't find is the dynamic loader.

E.g. try on host `file /bin/ls`, which gives `interpreter /lib64/ld-linux-x86-64.so.2,`.

lvcargnini, 2018-04-10:
As you can see no, but I let it compile using GCC, will revert to clang and create a static binary, thanks.

# ldd /usr/bin/memtest
[  107.007415] serial8250: too much work for irq4
ldd /usr/bin/memtest
checking sub-depends for 'not found'
checking sub-depends for 'not found'
checking sub-depends for '/usr/lib/libstdc++.so.6'
checking sub-depends for '/lib64/libgcc_s.so.1'
checking sub-depends for 'not found'
checking sub-depends for '/lib64/libc.so.0'
	ld64-uClibc.so.1 => /lib/ld64-uClibc.so.1 (0x7ffff7df7000)
	libpthread.so.0 => not found (0x00000000)
	librt.so.1 => not found (0x00000000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00000000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00000000)
	libc.so.6 => not found (0x00000000)
	libc.so.0 => /lib64/libc.so.0 (0x00000000)
	/lib/ld64-uClibc.so.1 => /lib/ld64-uClibc.so.1 (0x00000000)
	/lib/ld64-uClibc.so.1 => /lib/ld64-uClibc.so.1 (0x00000000)
# 
#

cirosantilli, 2020-01-09:
BTW, I also created a dedicated section for this at: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/657c59249e56d861bb0a437a1b1c757797281910#add-new-files-to-the-buildroot-image