cirosantilli/linux-kernel-module-cheat #23: fatal error upon build

kamomil, 2018-04-19:
Hi,
I have ubuntu 16.04
I run `./configure` and then `./build`
few seconds after the `./build` I get:

```
ui/input-keymap.c:8:44: fatal error: ui/input-keymap-linux-to-qcode.c: No such file or directory
```

cirosantilli, 2018-04-19:
Hi Dafna,

It looks like QEMU did not properly clone all its submodules (`ui/keycodemapdb`).

Try:

```
git submodule update --init --recursive
```

I would also be interested to know if there are earlier error messages to understand why the clone failed.

cirosantilli, 2018-04-19:
Also possibly fixed by: https://github.com/cirosantilli/linux-kernel-module-cheat/commit/3fdabbda61a8f0a5d97df13f204008a84cc3e6a2

kamomil, 2018-04-19:
Hi, I ran as you suggested and got:

```
dafna@guri:~/outreachy/linux-kernel-module-cheat$ git submodule update --init --recursive
Submodule 'gem5/gem5' (https://gem5.googlesource.com/public/gem5) registered for path 'gem5/gem5'
Submodule 'parsec-benchmark/parsec-benchmark' (https://github.com/cirosantilli/parsec-benchmark) registered for path 'parsec-benchmark/parsec-benchmark'
Cloning into 'gem5/gem5'...
remote: Sending approximately 159.38 MiB ...
remote: Counting objects: 52, done
remote: Finding sources: 100% (52/52)
remote: Total 177334 (delta 139261), reused 177321 (delta 139261)
Receiving objects: 100% (177334/177334), 159.25 MiB | 1.30 MiB/s, done.
Resolving deltas: 100% (139261/139261), done.
Checking connectivity... done.
Submodule path 'gem5/gem5': checked out '68af229490fc811aebddf68b3e2e09e63a5fa475'
Cloning into 'parsec-benchmark/parsec-benchmark'...
remote: Counting objects: 19805, done.
remote: Compressing objects: 100% (267/267), done.
remote: Total 19805 (delta 144), reused 253 (delta 102), pack-reused 19429
Receiving objects: 100% (19805/19805), 42.91 MiB | 1.02 MiB/s, done.
Resolving deltas: 100% (5012/5012), done.
Checking connectivity... done.
fatal: reference is not a tree: 580f134dc6d753531bf676b07c574d6b28b70e1d
Submodule 'capstone' (git://git.qemu.org/capstone.git) registered for path 'capstone'
Submodule 'dtc' (git://git.qemu-project.org/dtc.git) registered for path 'dtc'
Submodule 'roms/QemuMacDrivers' (git://git.qemu.org/QemuMacDrivers.git) registered for path 'roms/QemuMacDrivers'
Submodule 'roms/SLOF' (git://git.qemu-project.org/SLOF.git) registered for path 'roms/SLOF'
Submodule 'roms/ipxe' (git://git.qemu-project.org/ipxe.git) registered for path 'roms/ipxe'
Submodule 'roms/openbios' (git://git.qemu-project.org/openbios.git) registered for path 'roms/openbios'
Submodule 'roms/openhackware' (git://git.qemu-project.org/openhackware.git) registered for path 'roms/openhackware'
Submodule 'roms/qemu-palcode' (git://github.com/rth7680/qemu-palcode.git) registered for path 'roms/qemu-palcode'
Submodule 'roms/seabios' (git://git.qemu-project.org/seabios.git/) registered for path 'roms/seabios'
Submodule 'roms/sgabios' (git://git.qemu-project.org/sgabios.git) registered for path 'roms/sgabios'
Submodule 'roms/skiboot' (git://git.qemu.org/skiboot.git) registered for path 'roms/skiboot'
Submodule 'roms/u-boot' (git://git.qemu-project.org/u-boot.git) registered for path 'roms/u-boot'
Submodule 'roms/vgabios' (git://git.qemu-project.org/vgabios.git/) registered for path 'roms/vgabios'
Submodule 'ui/keycodemapdb' (git://git.qemu.org/keycodemapdb.git) registered for path 'ui/keycodemapdb'
Cloning into 'capstone'...
remote: Counting objects: 22832, done.
remote: Compressing objects: 100% (6683/6683), done.
remote: Total 22832 (delta 16520), reused 21860 (delta 15769)
Receiving objects: 100% (22832/22832), 32.70 MiB | 1.05 MiB/s, done.
Resolving deltas: 100% (16520/16520), done.
Checking connectivity... done.
Submodule path 'qemu/capstone': checked out '22ead3e0bfdb87516656453336160e0a37b066bf'
Cloning into 'dtc'...
remote: Counting objects: 4271, done.
remote: Compressing objects: 100% (1658/1658), done.
remote: Total 4271 (delta 3151), reused 3450 (delta 2571)
Receiving objects: 100% (4271/4271), 965.94 KiB | 623.00 KiB/s, done.
Resolving deltas: 100% (3151/3151), done.
Checking connectivity... done.
Submodule path 'qemu/dtc': checked out '558cd81bdd432769b59bff01240c44f82cfb1a9d'
Cloning into 'roms/QemuMacDrivers'...
remote: Counting objects: 60, done.
remote: Compressing objects: 100% (59/59), done.
remote: Total 60 (delta 24), reused 0 (delta 0)
Receiving objects: 100% (60/60), 66.37 KiB | 0 bytes/s, done.
Resolving deltas: 100% (24/24), done.
Checking connectivity... done.
Submodule path 'qemu/roms/QemuMacDrivers': checked out 'd4e7d7ac663fcb55f1b93575445fcbca372f17a7'
Cloning into 'roms/SLOF'...
remote: Counting objects: 6706, done.
remote: Compressing objects: 100% (1966/1966), done.
remote: Total 6706 (delta 5144), reused 6089 (delta 4679)
Receiving objects: 100% (6706/6706), 2.03 MiB | 575.00 KiB/s, done.
Resolving deltas: 100% (5144/5144), done.
Checking connectivity... done.
Submodule path 'qemu/roms/SLOF': checked out '89f519f09bf850918b60526e50409afb663418aa'
Cloning into 'roms/ipxe'...
[... 80 lines cut ...]

now when running `./build` I get

```
ui/input-keymap.c:8:44: fatal error: ui/input-keymap-linux-to-qcode.c: No such file or directory
```

I get some messages at the buggining of the `./build` instruction:

```
dafna@guri:~/outreachy/linux-kernel-module-cheat$ ./build
patching file package/qemu/qemu.mk
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored
patching file package/flex/0001-flex-disable-documentation.patch
Reversed (or previously applied) patch detected!  Skipping patch.
3 out of 3 hunks ignored
The next patch would create the file package/flex/0002-build-AC_USE_SYSTEM_EXTENSIONS-in-configure.ac.patch,
which already exists!  Skipping patch.
1 out of 1 hunk ignored
patching file package/flex/flex.mk
Hunk #1 FAILED at 10.
1 out of 1 hunk FAILED

```

cirosantilli, 2018-04-19:
The line:

```
Unable to checkout '580f134dc6d753531bf676b07c574d6b28b70e1d' in submodule path 'parsec-benchmark/parsec-benchmark'
```

indicates that you are on an old commit, the parsec submodule is not at that commit anymore.

Please pull this repo to `master` at e037ddcef5532f7b684f6bed2809bc2be3aa1bbf and try again.

kamomil, 2018-04-19:
hi,
the command `git submodule update --depth 1 $gitjobs --init -- $submodules` in the `configure` file is wrong.
I get a 'usage' message for it.

kamomil, 2018-04-19:
seems like my git version (2.7.4)  does not support the `--jobs` argument

cirosantilli, 2018-04-19:
The intention was that this should not be a problem in theory: errors there should jump to: https://github.com/cirosantilli/linux-kernel-module-cheat/blob/2852fe1989a6f1ab546e9a4fa88724423b3949f5/configure#L113

But maybe there is an error with the fallback mechanism.

kamomil, 2018-04-19:
ok, now the git command returns with no errors and yet the `./build` return with
```
ui/input-keymap.c:8:44: fatal error: ui/input-keymap-linux-to-qcode.c: No such file or directory
```

cirosantilli, 2018-04-19:
Are there any files under:

```
ls qemu/ui/keycodemapdb
```

?

kamomil, 2018-04-19:
```
dafna@guri:~/outreachy/linux-kernel-module-cheat/qemu$ ls ui/keycodemapdb/
data  LICENSE.BSD  LICENSE.GPL2  README  tests  thirdparty  tools
```
```
dafna@guri:~/outreachy/linux-kernel-module-cheat/qemu$ cd ui/keycodemapdb/
dafna@guri:~/outreachy/linux-kernel-module-cheat/qemu/ui/keycodemapdb$ git status
HEAD detached at 10739aa
nothing to commit, working directory clean
```

cirosantilli, 2018-04-19:
OK, now I am out of ideas, besides recommending that you nuke everything and restart a clean build :-)

seisdr, 2021-08-08:
```
CC      ui/input-keymap.o
/root/qemu-hangover-0.6.5/ui/input-keymap.c:7:10: fatal error: ui/input-keymap-atset1-to-qcode.c: No such file or directory
    7 | #include "ui/input-keymap-atset1-to-qcode.c"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
```