cirosantilli/linux-kernel-module-cheat #24: git?

cewatkins, 2018-04-24:
Unable to checkout '8ce27bb9fee80a406a4199657ef90e3c315e7457' in submodule path 'buildroot'
Unable to checkout '8dbff6d3728ff78c4da74d667bb4833edade06aa' in submodule path 'qemu'
I know using submodules is a pita; but I tried quite a few nasty hacks & couldn't get the docker
method working.. I am curious if it's just the simple commit issue..

cirosantilli, 2018-04-24:
What is your git and OS version?

I can't reproduce locally on master https://github.com/cirosantilli/linux-kernel-module-cheat/tree/14965a40d27c8d9d1ff5b023ace827b288a024ef , and the commits do seem present on remotes:

- 8ce27bb9fee80a406a4199657ef90e3c315e7457 https://github.com/buildroot/buildroot/commit/8ce27bb9fee80a406a4199657ef90e3c315e7457
- 8dbff6d3728ff78c4da74d667bb4833edade06aa  https://github.com/cirosantilli/qemu/tree/8dbff6d3728ff78c4da74d667bb4833edade06aa

Also give the Docker setup a try: https://github.com/cirosantilli/linux-kernel-module-cheat#docker

cewatkins, 2018-04-24:
Im on ubuntu 17.10, git is 2.14.1,
they checkout fine inside the os, if I try to repeat the steps as the docs suggest, it gets
very pick about which modules it's checking out, I tried separate trees,
so like i said it's inside docker like suggested it fails, I have it otherwise working on a few vms,
but the way the package works, I wanted to be able to completely yank the footprint, therefor docker.
It's looking pretty cool, especially since mips doesn't work  well with libvirt..

cirosantilli, 2018-04-24:
Hmm, I'm also on 17.10. Let me know if you find a solution. `./configure` is designed to minimize large clones, but direct `submodule update` should work fine too.

cewatkins, 2018-04-24:
Yes, like I said, it just doesnt work in docker, I tried a few images, if you do ./configure from docker,
it gets a different hash than one gets on the machine directly co tree, that's in the post,
I couldnt find them but they have to be in the tree somewhere, it was consistent; host pulled proper hash, docker pulled different identical hash, tried from several configs.
Im not a git guy at all, along with your notes I used separate clone's, same clone, etc.

cewatkins, 2018-04-24:
opps