cirosantilli/linux-kernel-module-cheat #30: Error running gem5: cannot find rootfs.ext2 on path

rivasjm, 2018-05-23:
Hi 

First of all, you have some amazing repositories here, thank you very much for all your work. I have a problem trying to run gem5 using your script. I use the command  `./run -a aarch64 -g`, but I get this error:

`IOError: Can't find file '/home/rivasjm/git/linux-kernel-modulecheat/out/aarch64/buildroot/images/rootfs.ext2' on path.`

Any ideas? I am using a clean ubuntu 18.04

cirosantilli, 2018-05-23:
Hi Juan,

What is your commit?

Did you build with:

```
./configure -g && ./build -a aarch64 -g
```

?

What is the output of:

```
ls -al out/aarch64/buildroot/images/
```

?

cirosantilli, 2018-05-23:
I have just done a clean build on master 3d0a766534d1e4ba71be0cf05f43a52bcaf6f594 and it worked here.

rivasjm, 2018-05-24:
Yes, I built with those commands before the run.

I ran previously the QEMU configuration with `./configure && ./build && ./run` and it worked. I then did the gem5 configuration in the same folder, and that's when I got that error message.

I was in commit 82872f9. I will do a clean installation with the latest commit now again. There is a chance there was a problem when I ran the configure or the build, I am running on a virtual machine and the host suspended a few times during that process.

rivasjm, 2018-05-24:
It works now. I started from scratch with a new VM, and just run the gem5 configuration/build. Thanks!

cirosantilli, 2018-05-24:
OK! I'll be closing this issue then, if you find any more problems, do open a new one.

closed 2018-05-24 by cirosantilli