cirosantilli/linux-kernel-module-cheat #32: Hi, how can I clone gem5 from githubmirror instead of googlesource

maokelong, 2018-09-11:
For some reason you might know, I can't access googlesource on my server. 

I tried to directly modify `.gitmodules` and sync, but things did't work, reporting errors like "Server does not allow request for unadvertised object".

cirosantilli, 2018-09-11:
@maokelong I'm innocent this time, I swear! :-)

Maybe it is due the attempt to `--shallow` clone? Did you try submodule update without it

Other possible alternative is to just cd into gem5 and `git remote add`, etc.

If all fails, go for https://stackoverflow.com/questions/913701/how-to-change-the-remote-repository-for-a-git-submodule and then ask a separate minimal question on SO if nothing there helps.

I've checked that the commit is there: https://github.com/gem5/gem5/commit/200281b under master.

maokelong, 2018-09-12:
@cirosantilli Thank you for your advise! I can run gem5 now~! :)

I did't try to remove the `shallow` option and the second method you provided seems didn't work because it still try to clone gem5 from the previous url .

What I did is to open `.git/config` and set `https://gem5.googlesource.com/public/gem5` to `https://github.com/gem5/gem5/`. Things then magically worked. 

Actually I do think there is no difference between those two attempts, but thanks god it works and I can run gem5 with the help of your magic cheat table now.

closed 2018-09-12 by maokelong

cirosantilli, 2018-09-12:
Great news! Let me know if you face further difficulties.