cirosantilli/linux-kernel-module-cheat #33: Writing custom kernel modules

willcray, 2018-09-13:
Hello, I've successfully booted and run your scripts, and they are awesome! I'm having a hard time figuring out how to build and test my own kernel modules in the QEMU system. For instance, if I want to modify hello.c to print something different upon doing insmod /hello.ko in the QEMU system or if I want to add my own kernel module, how do I got about doing that? This could be a nice addition to the documentation.

Thanks for your hard work!

cirosantilli, 2018-09-13:
Hey Will,

Just hack up the module source file https://github.com/cirosantilli/linux-kernel-module-cheat/blob/83b36867cf06ffdca3ce04296a8568d4f37ea13b/kernel_module/hello.c with your new message.

Then rebuild and reboot. `-k` is needed as explained at: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/83b36867cf06ffdca3ce04296a8568d4f37ea13b#rebuild

````
./build -k
./run
````

BTW, I'm going to be putting out a large rework soon, currently on a branch, and I gave a bit more emphasis to that subject: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/3d978570bcf667ce059d675a418cc0c62966813b#your-first-kernel-hack

Let me know if you have any further problems.

willcray, 2018-09-14:
Thanks! Knowing to use the -k flag would've saved me a couple of hours yesterday :). I appreciate the quick response!