cirosantilli/linux-kernel-module-cheat #35: Ubuntu image

cirosantilli, 2018-09-25:
Compile kernel separately from Buildroot.

Download docker image and convert it to regular image.

andrewzigerelli, 2019-01-23:
Hi Ciro,

Do you have any updates on this?

I can boot ubuntu 16.04.5, but I cannot for the life of me figure out how to modify gem5's init script to work with systemd.

cirosantilli, 2019-01-23:
Hi andrew,

I ended up focusing on other features and didn't get to this.

What did you try, and what problem did you observe? I'm not a systemd expert, so I might learn from your attempt.

Can't you just mount the image on host, and add your gem5 script run at the end of boot with one of: https://askubuntu.com/questions/814/how-to-run-scripts-on-start-up

Then test in QEMU to make testing loop faster before going to gem5.

andrewzigerelli, 2019-01-24:
Thanks for the link, it works!

I used the "For 15.04 and later:" answer; I'm booting 16.04 with kernel 4.19.0

I haven't yet figured out why the default gem5.service (provided by gem5-guest-tools on Jason's blog) doesn't work. I don't know if it was because I removed the After=getty.target, changed Type to oneshot, or changed WantedBy=multi-user.target.

I think it should be obvious if I dig into how systemd actually works.

TODO: Figure out why this worked.
TODO: Modify it to use your gem5.sh instead (need to understand --eval-after flag and if it applies)
TODO: Auto set this up via https://github.com/jimmysitu/gem5-bench