cirosantilli/linux-kernel-module-cheat #36: Fatal Error: #include <linux/uaccess.h> no such file or directory

willcray, 2018-09-27:
I'm getting this error while trying to construct an ioctl interface between a userspace app in /packages/kernel-modules/user and a custom device driver in  /packages/kernel-modules/. I'm trying to communicate between the two via copy_to_user(). The app and device driver worked in this setup before I included the copy to user file header and function.

What's interesting is that your ioctl.c and icotl.h use this functionality, and I can't figure out how to extend that functionality to my kernel module and app. I'm running 4.18.0. Is it because you don't have the 4.18.0 linux headers in buildroot?

cirosantilli, 2018-09-27:
Hey Will,

Is `#include <linux/uaccess.h>` in the kernel module or in the user?

I don't see how it can fail on kernel module, since my examples use it.

For user, I have never used it, so I'm not sure. Why would you want to include that file, when all I needed was the POSIX `#include <sys/ioctl.h>` + `#include "../ioctl.h"` which contains only the kernel module custom defined struct and magic? https://github.com/cirosantilli/linux-kernel-module-cheat/blob/6b563c0234de948da56df9924e0cf4e671cf220c/packages/kernel_modules/ioctl.h#L7

Also provide a minimal branch or patch so I can try to reproduce if the problem persists.

willcray, 2018-09-27:
Cleaning the user directory fixed it! Although, I had to go all the way into `/out/buildroot/x86_64/default/build/kernel_modules-1.0/user` and remove the files containing the `#include <linux/uaccess.h>` line. Residual files in `/out/buildroot/x86_64/default/build/kernel_modules-1.0/ and `/out/buildroot/x86_64/default/build/kernel_modules-1.0/user` kernel modules folder has been a persistent issue, despite removing the files from the `/packages/kernel_modules` and `/packages/kernel_modules/user` space and rebuilding with `./build-buildroot -k`.

willcray, 2018-09-27:
Cleaning the user directory fixed it! Although, I had to go all the way into `/out/buildroot/x86_64/default/build/kernel_modules-1.0/user` and remove the files containing the `#include <linux/uaccess.h>` line. Residual files in `/out/buildroot/x86_64/default/build/kernel_modules-1.0/` and `/out/buildroot/x86_64/default/build/kernel_modules-1.0/user` kernel modules folder has been a persistent issue, despite removing the files from the `/packages/kernel_modules` and `/packages/kernel_modules/userspace` and rebuilding with `./build-buildroot -k`.

cirosantilli, 2018-09-28:
OK.

Yes, as you've learnt the hard way, Buildroot does not clear the build directory, see also: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/d1a9b8e11cf99d1cb7e2bc9e3b3865f322390739#clean-the-build

I intend to support buliding the kernel modules outside of the Buildroot build system, and to use with 9p at some point, that will make things much nicer.

closed 2018-09-28 by cirosantilli