cirosantilli/linux-kernel-module-cheat #39: Downgrade Kernel version, build failed

reveriel, 2018-11-13:
## in short
Try to use 3.18 linux kernel, failed at `./build-buildroot`,
`BR2_EXTERNAL` has strange value.  Setting it to empty avoids it.

Is there something that are not cleaned? 

## details
I need to test the ver3.18 kernel. 
So I checked out ver3.18 in `submodule/linux`
and checked out the corresponding buildroot commit as the readme said. 
then I run `./build`, and failed in `./build-buildroot`
```
± % ./build-buildroot
+ cd /home/gx/linux-kernel-module-cheat/submodules/buildroot && \
  make \
  O=/home/gx/linux-kernel-module-cheat/out/buildroot/build/default/x86_64 \
  BR2_EXTERNAL=../../buildroot_packages/parsec_benchmark:../../buildroot_packages/sample_package:../../buildroot_packages/lkmc_many_files:../../buildroot_packages/kernel_modules \
  qemu_x86_64_defconfig \
;
Makefile:140: *** BR2_EXTERNAL='../../buildroot_packages/parsec_benchmark:../../buildroot_packages/sample_package:../../buildroot_packages/lkmc_many_files:../../buildroot_packages/kernel_modules' does not exist, relative to /home/gx/linux-kernel-module-cheat/submodules/buildroot.  Stop.
Traceback (most recent call last):
  File "./build-buildroot", line 188, in <module>
    BuildrootComponent().build()
  File "/home/gx/linux-kernel-module-cheat/common.py", line 111, in build
    self.do_build(args)
  File "./build-buildroot", line 95, in do_build
    cwd=common.buildroot_src_dir,
  File "/home/gx/linux-kernel-module-cheat/common.py", line 683, in run_cmd
    raise Exception('Command exited with status: {}'.format(returncode))
Exception: Command exited with status: 2
```

And the related code is  `commit 863867ed3d9079e7b820b144724d81fc1becec31`
 `submodule/buildroot/Makefile:140`

``` makefile
BR2_EXTERNAL_FILE = $(BASE_DIR)/.br-external
-include $(BR2_EXTERNAL_FILE)
ifeq ($(BR2_EXTERNAL),)
  override BR2_EXTERNAL = support/dummy-external
  $(shell rm -f $(BR2_EXTERNAL_FILE))
else
  _BR2_EXTERNAL = $(shell cd $(BR2_EXTERNAL) >/dev/null 2>&1 && pwd)
  ifeq ($(_BR2_EXTERNAL),)
    $(error BR2_EXTERNAL='$(BR2_EXTERNAL)' does not exist, relative to $(TOPDIR))
  endif
  override BR2_EXTERNAL := $(_BR2_EXTERNAL)
  $(shell echo BR2_EXTERNAL ?= $(BR2_EXTERNAL) > $(BR2_EXTERNAL_FILE))
endif
```

this code tries to cd `$(BR2_EXTERNAL)` and `pwd` , but the value of `BR2_EXTERNAL` is such a long string that contains many path separated by `:`,  I don't know where is the definition of `BR2_EXTERNAL` is

So I add 
``` makefile
override BR2_EXTERNAL =
```
and the build continues..

cirosantilli, 2018-11-13:
Hi Reveriel,

The current Buildroot allows BR2_EXTERNAL to be a colon : separated list of paths: https://buildroot.org/downloads/manual/manual.html#outside-br-custom

So I'm guessing that the Buildroot you checked out to did not support that? Which version was that for the record?

In general, downgrading the kernel is a pain, since you have to mess with the Buildroot version, and then potentially you also have to make this repo play well with that version of Buildroot. I've updated the https://github.com/cirosantilli/linux-kernel-module-cheat/tree/7bbc3c4c8bb556267aeeb820461c7cf99c6b9d79#downgrade-the-linux-kernel section to clarify that now.

For the most part however, I expect the required changes to this repo to be trivial: just do a `--dry-run` build to see what `./build-buildroot` is doing, then build the Buildroot docs for that buildroot version to see which options you need, and then grep our scripts to see what is setting those options and hack it up.

Another thing you can try is to run an older kernel on our current Buildroot. It might work for the most part, but I wouldn't do any serious work based on it.

If this is important to you and if you understand what is going on, send me a pull request with a  linux-3.18 branch of this repo and I'll let you own it.

reveriel, 2018-11-14:
The buildroot version is 
```
commit 863867ed3d9079e7b820b144724d81fc1becec31 (HEAD)
Author: Gustavo Zacarias <gustavo@zacarias.com.ar>
Date:   Tue Dec 16 15:52:18 2014 -0300

    linux: bump default to version 3.18.1
```
The default version is 3.18.1 not 3.18 beacause 3.18 failed to compile at `host-binutils`.

the 3.18.1 buildroot's manual said:
```
    The main usage of this is to store package recipes. The recommended way to do this is to write a $(BR2_EXTERNAL)/Config.in file that looks like:

    source "$BR2_EXTERNAL/package/package1/Config.in"
    source "$BR2_EXTERNAL/package/package2/Config.in"

    Then, have a $(BR2_EXTERNAL)/external.mk file that looks like:

    include $(sort $(wildcard $(BR2_EXTERNAL)/package/*/*.mk))

    And then in $(BR2_EXTERNAL)/package/package1 and $(BR2_EXTERNAL)/package/package2 create normal Buildroot packag
```
It seems we need to add a `Config.in` and `externe.ml` to include multiple external packages. 

There are some problems when building the 3.18.1 buildroot.
1. `host-gcc-final-4.8.3` build failed. Need to apply a [patch](https://gcc.gnu.org/git/?p=gcc.git;a=commitdiff;h=ec1cc0263f156f70693a62cf17b254a0029f4852)
2. ` host-ncurses-5.9` build failed. I add "-P" to the `CPPFLAGS` in `/host-ncurses-5.9/ncurses/Makefile`, then make clean. [ref](https://stackoverflow.com/questions/37475222/ncurses-6-0-compilation-error-error-expected-before-int)
3. Need `mkpasswd` which is not included by default in Ubt18.04LTS. `apt install whois`

Your lkmc helps me a lot, and I'm willing to contribute. But I'm quite new to kernel development, and am not clear what buildroot does. As for those external packages, right now I just ignored them. 

Thanks for your explanation.

closed 2018-11-14 by reveriel

cirosantilli, 2018-11-14:
OK, glad you managed to find out what happened.

If something else goes wrong, let me know.

There is no obligation to contribute, as long as it helped you, I'm happy.

Buildroot are a way to add your software to the Buildroot image, and `BR2_EXTERNAL` is required for packages outside the Buildroot source tree.

All our Buildroot packages are under buildroot_packages/ to the build. But there is nothing important there anymore, as I moved the important stuff to custom build commands like ./build-modules and ./build-userland which integrate with `BR2_ROOTFS_OVERLAY`.

More info at: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/f06e26f865fcc87d630f7766415c303bcf202f29#buildroot_packages-directory