cirosantilli/linux-kernel-module-cheat #41: Is Ubutun 16 LTS still supported?

9527KGG, 2018-11-19:
./build --download-dependencies qemu-buildroot
prompts:

+ sudo apt-get build-dep \
  qemu \
;
Reading package lists... Done
E: You must put some 'source' URIs in your sources.list
Traceback (most recent call last):
  File "./build", line 421, in <module>
    common.add_newlines(sorted(apt_build_deps))
  File "/home/k/src/lkc1/common.py", line 687, in run_cmd
    raise Exception('Command exited with status: {}'.format(returncode))
Exception: Command exited with status: 100

Thanks.

cirosantilli, 2018-11-19:
Yes, always try to grep readme and google for error message: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/454af5d03a45498d2c62a424071dc675ff59c4e0#you-must-put-some-source-uris-in-your-sources-list

9527KGG, 2018-11-19:
oh. thanks. working now.

closed 2018-11-19 by 9527KGG