cirosantilli/linux-kernel-module-cheat #44: Build fails with: "fatal: reference is not a tree:"

mmarb, 2018-11-26:
fatal: reference is not a tree: 84df9525b0c27f3ebc2ebb1864fa62a97fdedb7d
Unable to checkout '84df9525b0c27f3ebc2ebb1864fa62a97fdedb7d' in submodule path 'submodules/linux'

cirosantilli, 2018-11-26:
I can't reproduce on master a06872241b724a84fc2c3baac008867dfb0f1590 on Ubuntu 18.04 git 2.17.1.

Please provide details about your distro, your git revision, and try the Docker setup.

The raw command that has to work is:

    git submodule update --init --recursive --depth 1 submodules/linux

as can be seen from:

    ./build --download-dependencies qemu-buildroot --dry-run

mmarb, 2018-11-26:
git submodule update --init --recursive --depth 1 submodules/linux
fatal: reference is not a tree: 84df9525b0c27f3ebc2ebb1864fa62a97fdedb7d
Unable to checkout '84df9525b0c27f3ebc2ebb1864fa62a97fdedb7d' in submodule path 'submodules/linux'

I am running:
Ubuntu 16.04.5 LTS
git version 2.7.4

cirosantilli, 2018-11-26:
I have also tested on Ubuntu 16.04 same git and it worked, so I'm confused :-)

Try removing `--depth 1`. In last case, just clone the kernel normally, and checkout to the right commit.

mmarb, 2018-11-26:
Removing `--depth 1` didn't help. However running the docker setup worked just fine.

ysun, 2018-12-03:
Hi Cirosantilli,

I encountered the same issue, the error info is as following:
Requirement already satisfied (use --upgrade to upgrade): pexpect==4.6.0 in /root/.local/lib/python3.5/site-packages                  
Requirement already satisfied (use --upgrade to upgrade): ptyprocess>=0.5 in /usr/lib/python3/dist-packages (from pexpect==4.6.0)     
You are using pip version 8.1.1, however version 18.1 is available.                                                                   
You should consider upgrading via the 'pip install --upgrade pip' command.                                                            
+ git submodule update --init --recursive -- \                                                                                        
  /home/works/linux-kernel-module-cheat/submodules/buildroot \                                                                        
  /home/works/linux-kernel-module-cheat/submodules/qemu \                                                                             
;                                                                                                                                     
+ git submodule update --init --recursive --depth 1 -- \                                                                              
  /home/works/linux-kernel-module-cheat/submodules/linux \                                                                            
;                                                                                                                                     
fatal: reference is not a tree: 84df9525b0c27f3ebc2ebb1864fa62a97fdedb7d                                                              
Unable to checkout '84df9525b0c27f3ebc2ebb1864fa62a97fdedb7d' in submodule path 'submodules/linux'                                    
Traceback (most recent call last):                                                                                                    
  File "./build", line 477, in <module>                                                                                               
    common.add_newlines([os.path.join(common.submodules_dir, x) for x in sorted(submodules_shallow)])                                 
  File "/home/works/linux-kernel-module-cheat/common.py", line 701, in run_cmd                                                        
    raise Exception('Command exited with status: {}'.format(returncode))                                                              
Exception: Command exited with status: 1          

There is no sha1 84df9525b0c27f3ebc2ebb1864fa62a97fdedb7d in repo submodules/linux, why the script was trying to checkout such commit ?
I'm using Ubuntu 16.04

ysun, 2018-12-03:
Ok, I figured it out. I found that missing sha1 is in linux-stable, then I copied it into submodule.
No error shown anymore.

cirosantilli, 2018-12-03:
@ysun thanks for the report,

I really don't understand why it is not working, since that commit does exist on the submodule tree directly under a branch: https://github.com/cirosantilli/linux/tree/lkmc

I wish I knew why I can't reproduce it :-(

If anyone finds out, let me know.

stiticrobert, 2018-12-26:
to fix cd into submodules/linux and git checkout 84df9525b0c27f3ebc2ebb1864fa62a97fdedb7d
after that cd .. twice, git add submodules/linux and git commit

cirosantilli, 2018-12-26:
@stiticrobert do you mean me? :-) Because it is already at that commit: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/3b0a343647bed577586989fb702b760bd280844a/submodules

cirosantilli, 2019-05-10:
I've updated the title following activity from: https://github.com/cirosantilli/linux-kernel-module-cheat/issues/62

cirosantilli, 2019-05-18:
I am closing this issue because I feel I fully understand it, and have worked around it in the best way I could find as summarized at: https://github.com/cirosantilli/linux-kernel-module-cheat/issues/44#issuecomment-441600370

cc. @ashriram as this might be of interest to you.

closed 2019-05-18 by cirosantilli