cirosantilli/linux-kernel-module-cheat #46: Parsec input size

andrewzigerelli, 2018-12-10:
```
./build-buildroot \
  --arch x86_64 \
  --config 'BR2_PACKAGE_PARSEC_BENCHMARK=y' \
  --config 'BR2_PACKAGE_PARSEC_BENCHMARK_INPUT_SIZE="simsmall"' \
  -- parsec-benchmark-reconfigure \
;

```
fails during the build with:

make[1]: *** No rule to make target 'parsec-benchmark-reconfigure'.  Stop.
Makefile:79: recipe for target '_all' failed

Further, do you have any advice about the rest of parsec that doesn't compile? I know I could just bite the bullet and use the disk image from UT, but I really like your setup and am willing to put in some time to make more benchmarks works.

Thanks!

cirosantilli, 2018-12-10:
Hey,

See if this fixed the problem: https://github.com/cirosantilli/linux-kernel-module-cheat/commit/9f6ddbc436f344dd77dd9d5646dbe643f97533e4

I debugged it by looking at what `sample_package` is doing and comparing them.

For the missing benchmarks: I have summarized my investigation of missing parts at: https://github.com/cirosantilli/parsec-benchmark/tree/75d55ac446a43c47efb1044844a108c6c330184c

But basically, there are two types of problems:

- smaller internal dependencies that just need manual porting, e.g. atomics for aarch64
- large external depencies that need small porting patches. e.g. cmake stuff. For those, I would just throw away the parsec stuff, and use as much of the pre packaged Ubuntu / Buildroot guest packaging system as possible.

    Then when you reach the software that has the ROIs, upgrade to the smae same version as the guest package, make compilation from source work, and then re-introduce the ROIs.

    Then add those hacked packages as proper git submodules of: https://github.com/cirosantilli/parsec-benchmark

     This leads to a more different benchmark, but who cares about ultra old software anyways?

andrewzigerelli, 2018-12-10:
Thank you very much! I should have made my grep for parsec-benchmark more general...

closed 2018-12-10 by andrewzigerelli