cirosantilli/linux-kernel-module-cheat #55: Build failure on AARCH64: /out/buildroot/build/default/aarch64/build/glibc-custom/configure: No such file or directory

guojiun, 2019-04-25:
I run the following command : 
`./build --arch aarch64 --download-dependencies qemu-buildroot`  

After it finished downloading submodules, it started building the tools for aarch64 platform.
But it got an error: 

>  /home/guojiun/sysprog21/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/glibc-custom/configure: No such file or directory

And surprisingly, I also found  /home/guojiun/sysprog21/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/glibc-custom/glibc-custom/build is empty.

I have no idea what happened.

cirosantilli, 2019-04-26:
Hi Guojin,

It looks like the glibc submoule under submodules/glibc did not clone for some reason. Buildroot copies that into the `out/buildroot/build/default/aarch64/build/glibc-custom` folder before building.

That submodule should be checked out automatically by `./build --download-dependencies`.

Can you try to `rm -rf out/buildroot/build/default/aarch64/build/glibc-custom` and then manually update the submodule with `git submodule update --init` and then try to build again?

I've tested a build from scratch at commit: https://github.com/cirosantilli/linux-kernel-module-cheat/commit/4943c9ed2ed63856b461cf3d883c26bee0b36df6 (note that I pushed something yesterday, but I don't think it mattered) and it did worked fine.

Another thing to try out is to checkout tag v3.0 which is the latest one, as the tags have been better tested.

gipi, 2019-11-08:
Happened the same to me and I can confirm that manually "fixing" the submodule makes the error disappear

cirosantilli, 2019-11-08:
@gipi thanks for the report, I would like to understand why this is happening, but I can't reproduce.. let me know if anyone finds anything.

gipi, 2019-11-09:
In my case was caused by me stopping the process midway to move the repository in a partition of a SSD disk, so I think it's not a bug from your part.