cirosantilli/linux-kernel-module-cheat #62: submodule need to be Fwd

ashriram, 2019-05-10:
How to replicate
Ubuntu 16.04
git clone lkmc
./build --download-dependencies --no-apt

===================================================================
...............
 git submodule update --init --recursive --depth 1 -- \
  /local-scratch/home/ashriram/build/Linux/native/linux-kernel-module-cheat/submodules/binutils-gdb \
  /local-scratch/home/ashriram/build/Linux/native/linux-kernel-module-cheat/submodules/gcc \
  /local-scratch/home/ashriram/build/Linux/native/linux-kernel-module-cheat/submodules/glibc \
  /local-scratch/home/ashriram/build/Linux/native/linux-kernel-module-cheat/submodules/linux \
;
fatal: reference is not a tree: 1690de78a3603a3a0ab8487ef896e75871bb0737
fatal: reference is not a tree: 9046071dddb0a9ee7e30e5edf27cfc53b5b9c238
.........
===================================================================
linux had a similar issue, but that seemed to resolve itself when I change the 
https://github.com/cirosantilli/linux to https://github.com/torvalds/linux

cirosantilli, 2019-05-10:
Hi ashriram,

What is your commit for this repo?

Those are the trees for gdb-binutils and gcc which are missing, and there was a recent fix for those at 7a5ca339a356fff98c1537f0b2c5e960757924d8 | https://github.com/cirosantilli/linux-kernel-module-cheat/issues/58

Maybe you need to run `git submodule sync` after getting that latest commit as well: https://stackoverflow.com/questions/913701/how-to-change-the-remote-repository-for-a-git-submodule

You can see that with:

```
git ls-tree HEAD:./submodules/
```

which gives:

```
160000 commit 1690de78a3603a3a0ab8487ef896e75871bb0737  binutils-gdb
160000 commit ed60963595855e66ffc06a8a543cbb429c7ede03  boot-wrapper-aarch64
160000 commit 653eaa178bf7352791d988a8ee9db79c8bd7975e  buildroot
160000 commit d5900debd397b8909d9cafeb9a1093fb7a5dc6e6  crosstool-ng
160000 commit 9046071dddb0a9ee7e30e5edf27cfc53b5b9c238  gcc
160000 commit 9048ef0ffbf21bedb803b785fb68f83e95c04db8  gem5
160000 commit 6c99e37f6fb640a50a3113b2dbee5d5389843c1e  glibc
160000 commit 1c163f4c7b3f621efff9b28a47abb36f7378d783  linux
160000 commit 75d55ac446a43c47efb1044844a108c6c330184c  parsec-benchmark
160000 commit d0f9ab350f315aa3643791b0587d28b38bf3b18c  qemu
160000 commit 96cbd0893f783997caaf117e897d5fa8f2dc7b5f  xen
```

and both of those are have been uploaded and are under a branch like the others:

- https://github.com/cirosantilli/binutils-gdb/commits/lkmc-binutils-2_29_1
- https://github.com/cirosantilli/gcc/tree/lkmc-gcc-7_3_0-release

ashriram, 2019-05-10:
Hi ---

First of all; thank you for putting this great resource together. I am using this in a classroom setting to teach OS to undergrads.

My commit:
commit 7a5ca339a356fff98c1537f0b2c5e960757924d8
Author: Ciro Santilli 六四事件 法轮功 <ciro.santilli@gmail.com>
Date:   Sun May 5 00:00:00 2019 +0000

    submodules: use https for binutils-gdb and gcc. Fix #58

I did try syncing it; but it still throws the 
"fatal: reference is not a tree: 1690de78a3603a3a0ab8487ef896e75871bb0737
fatal: reference is not a tree: 9046071dddb0a9ee7e30e5edf27cfc53b5b9c238
Unable to checkout '1690de78a3603a3a0ab8487ef896e75871bb0737' in submodule path 'submodules/binutils-gdb'
Unable to checkout '9046071dddb0a9ee7e30e5edf27cfc53b5b9c238' in submodule path 'submodules/gcc'"

------------------------------------------------------------------------------------------------------

The weird part is your docker instructions work. It only throws an issue if I
git clone lkml
./build --download-dependencies --no-apt

Regards

ashriram, 2019-05-10:
Issue #58 was for fetching. The challenge here is that your fork of the binutils, gcc and linux seem to be behind the mainline

cirosantilli, 2019-05-10:
I'm glad you found this useful!!

Hmmm, are you doing a clean clone? I've just tested the commands on Ubuntu 19.04:

```
git clone https://github.com/cirosantilli/linux-kernel-module-cheat lkmc
./build --download-dependencies --no-apt
```

and it works.

Since it works on docker for you on Ubuntu 16.04, I'm starting to wonder if it is a git version problem then.

My fork if behind mainline, but by repo does include the hashes on branches as I've shown at:

- https://github.com/cirosantilli/binutils-gdb/commit/1690de78a3603a3a0ab8487ef896e75871bb0737
- https://github.com/cirosantilli/gcc/commit/9046071dddb0a9ee7e30e5edf27cfc53b5b9c238
- https://github.com/cirosantilli/linux/commit/1c163f4c7b3f621efff9b28a47abb36f7378d783

Can you try to compile latest git tag from master and use that to submodule update?

Otherwise I will try things on an Ubuntu 16.04 Docker later on

ashriram, 2019-05-10:
Bingo! 
It was a git version problem. 
Ubuntu 16.04 was running git 2.7 .
 I upgraded it to 2.21 and now thing seem to be ok.

To upgrade:
sudo add-apt-repository ppa:git-core/ppa -y
sudo apt-get update
sudo apt-get install git -y
git --version

cirosantilli, 2019-05-10:
Arghh, thanks for checking this out. I will update the title and keep this around open for now to see if we can find a set of commands that will work, I definitely want it to just work on supported all Ubuntu if that is not too hard.

cirosantilli, 2019-05-10:
Oh wait, I'll mark this as a dupe of https://github.com/cirosantilli/linux-kernel-module-cheat/issues/44 I had forgotten about that!

closed 2019-05-10 by cirosantilli