cirosantilli/linux-kernel-module-cheat #66: userland/.build error

DeadRabbits307, 2019-05-21:
uname -r: 4.19.32-1-MANJARO

Following Output appears running `userland/.build:`

```
linux-kernel-module-cheat-master_copy/userland/c/assert_fail.c:1:1: error: expected identifier or \u2018(\u2019 before \u2018.\u2019 token
 ../../lkmc/assert_fail.c
 ^
/linux-kernel-module-cheat-master_copy/userland/c/hello.c:1:1: error: expected identifier or \u2018(\u2019 before \u2018.\u2019 token
 ../../lkmc/hello.c
 ^
/usr/bin/ld: cannot find -lgomp
collect2: error: ld returned 1 exit status
/usr/bin/ld: cannot find -lgomp
collect2: error: ld returned 1 exit status
/usr/bin/ld: cannot find -lgomp
collect2: error: ld returned 1 exit status
Command exited with status: 1
time 00:00:00
```

Note: before I had to set softlinks:

`sudo ln -s /usr/bin/gcc /usr/bin/x86_64-linux-gnu-gcc 
sudo ln -s /usr/bin/g++ /usr/bin/x86_64-linux-gnu-g++`

Is something missing or the content of files like c/hello.c corrupted (".."?)

DeadRabbits307, 2019-05-21:
Problem was, unzipping the Git-Repo under Windows!

closed 2019-05-21 by DeadRabbits307

cirosantilli, 2019-05-21:
Thanks for letting me know that you've found the error! I was already starting to `hd` the file looking for weird bytes! Why are you unzipping the repo in window BTW? :-)

I have now removed the need for the symlink step at: https://github.com/cirosantilli/linux-kernel-module-cheat/commit/878eb0ad9837997247ce08479df080a06b035637

DeadRabbits307, 2019-05-21:
Because my setup consists of VS Code running on Windows and as terminal I use WSL with a running Arch Linux.  So the files were unpacked on a Windows filesystem.
To be honest, doing this step I didn't consider the softlink-incompatibility with Windows