cirosantilli/linux-kernel-module-cheat #84: make: *** [/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/glibc-custom/.stamp_configured] Error 127

guoqinglei, 2019-09-07:
tom@tom-pc:~/git/linux-kernel-module-cheat$ sudo -H ./build --download-dependencies gem5-buildroot 

the detailed error info is as follows:

 
```
   + sudo apt-get update
   Hit:1 http://mirrors.aliyun.com/ubuntu bionic InRelease
   Hit:2 http://mirrors.aliyun.com/ubuntu bionic-updates InRelease
   Hit:3 http://mirrors.aliyun.com/ubuntu bionic-backports InRelease
   Hit:4 http://mirrors.aliyun.com/ubuntu bionic-security InRelease
   Reading package lists... Done                      
   + sudo apt-get install \
     bash \
     bc \
     binutils \
     bison \
     build-essential \
     bzip2 \
     cpio \
     device-tree-compiler \
     diod \
     flex \
     g++ \
     g++-aarch64-linux-gnu \
     g++-arm-linux-gnueabihf \
     gcc \
     gcc-aarch64-linux-gnu \
     gcc-arm-linux-gnueabihf \
     git \
     graphviz \
     gzip \
     libboost-all-dev \
     libelf-dev \
     libgoogle-perftools-dev \
     m4 \
     make \
     moreutils \
     patch \
     perl \
     protobuf-compiler \
     python-dev \
     python-matplotlib \
     python-pip \
     python3 \
     python3-pip \
     qemu-utils \
     rr \
     rsync \
     scons \
     sed \
     tar \
     tmux \
     unzip \
     vinagre \
     wget \
     zlib1g-dev \
   ;
   Reading package lists... Done
   Building dependency tree       
   Reading state information... Done
   bc is already the newest version (1.07.1-2).
   bison is already the newest version (2:3.0.4.dfsg-1build1).
   build-essential is already the newest version (12.4ubuntu1).
   cpio is already the newest version (2.12+dfsg-6).
   device-tree-compiler is already the newest version (1.4.5-3).
   flex is already the newest version (2.6.4-6).
   gzip is already the newest version (1.6-5ubuntu1).
   libgoogle-perftools-dev is already the newest version (2.5-2.2ubuntu3).
   m4 is already the newest version (1.4.18-1).
[... 159 lines cut ...]
   ret = self.timed_main()
   File "./build", line 605, in timed_main
   component.build(self.env['arch'])
   File "./build", line 56, in build
   self.build_callback()
   File "./build", line 414, in f
   lkmc.import_path.import_path_main(component_file)(**args)
   File "/home/tom/git/linux-kernel-module-cheat/common.py", line 643, in __call__
   return super().__call__(**kwargs)
   File "/home/tom/git/linux-kernel-module-cheat/cli_function.py", line 149, in __call__
   return self._do_main(kwargs)
   File "/home/tom/git/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
   return self.main(**self._get_args(kwargs))
   File "/home/tom/git/linux-kernel-module-cheat/common.py", line 1289, in main
   ret = self.timed_main()
   File "/home/tom/git/linux-kernel-module-cheat/common.py", line 1712, in timed_main
   return self.build()
   File "/home/tom/git/linux-kernel-module-cheat/build-buildroot", line 166, in build
   cwd=self.env['buildroot_source_dir'],
   File "/home/tom/git/linux-kernel-module-cheat/shell_helpers.py", line 403, in run_cmd
   raise e
   Exception: Command exited with status: 2
```

anyone knows why this error happens ?

cirosantilli, 2019-09-07:
I'll try it out on my machine.

But in any case, please don't use `sudo` with these scripts! :-)

cirosantilli, 2019-09-07:
Hmmm, I cloned master at 6aa375df2ad3ce2e7d741d09b378503c25547df1 but `./build --download-dependencies gem5-buildroot` worked for me on my Ubuntu 19.04.

Can you try to nuke everything and restart from scratch at that commit, without sudo?

If that doesn't work still, give the Docker setup a try: https://github.com/cirosantilli/linux-kernel-module-cheat/tree/6aa375df2ad3ce2e7d741d09b378503c25547df1#docker

guoqinglei, 2019-09-08:
> I'll try it out on my machine.
> 
> But in any case, please don't use `sudo` with these scripts! :-)

I have pulled the master at 6aa3754 and run "./build --download-dependencies gem5-buildroot" without nuking everything from scratch, the error still exist. 

My ubuntu is 18.04 LTS, do I need to clean the build directory generated under "sudo ./build ***" ?
and do the ./build from the scratch ?

cirosantilli, 2019-09-08:
Yes, I recommend trying to completely remove the whole linux-kernel-module-cheat/ directory, then clone again and rebuild without sudo. I did a full clean build and it "worked on my machine" :-)

cirosantilli, 2019-09-08:
You can try to see the state of the buildroot build, but I don't think it is worth it at this point, just too much complication. If the clean build fails, then try it.

You have to understand why that happened:

```
out/buildroot/build/default/x86_64/build/glibc-custom/configure: No such file or directory
```

Most likely candidate was that the submodule/glibc was not cloned properly. But `./bulid` should do that automatically.

guoqinglei, 2019-09-08:
> You can try to see the state of the buildroot build, but I don't think it is worth it at this point, just too much complication. If the clean build fails, then try it.
> 
> You have to understand why that happened:
> 
> ```
> out/buildroot/build/default/x86_64/build/glibc-custom/configure: No such file or directory
> ```
> 
> Most likely candidate was that the submodule/glibc was not cloned properly. But `./build` should do that automatically.

I have cleaned the directory ~/git/linux-kernel-module-cheat and rebuild again but it stalls here:

usr/bin/make -j1 O=/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/x86_64 HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" silentoldconfig
make[1]: Entering directory '/home/tom/git/linux-kernel-module-cheat/submodules/buildroot'
  GEN     /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/Makefile
make[1]: Leaving directory '/home/tom/git/linux-kernel-module-cheat/submodules/buildroot'
>>> host-pkgconf 0.9.12 Downloading

It stalls here waiting " host-pkgconf 0.9.12 Downloading" for over half an hour. May the network speed is very slow. But it's strange that there is no progress bars to indicate.

guoqinglei, 2019-09-08:
> You can try to see the state of the buildroot build, but I don't think it is worth it at this point, just too much complication. If the clean build fails, then try it.
>
> You have to understand why [...]

i am so happy that the clean build has succeeded. Although it took a long time to go. 
the tail info is as follows:

Creating regular file /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/images/rootfs.ext2
Creating filesystem with 131072 4k blocks and 32768 inodes
Filesystem UUID: 454214f4-002e-45aa-91a9-7d62fbc26396
Superblock backups stored on blocks: 
	32768, 98304

Allocating group tables: done                            
Writing inode tables: done                            
Copying files into the device: done
Writing superblocks and filesystem accounting information: done

time 03:44:15

Thanks a lot.

cirosantilli, 2019-09-08:
Great!

Yes, the build is a bit slow, I get 30 minutes from scratch on my high end 2018 laptop: https://cirosantilli.com/linux-kernel-module-cheat/#benchmark-builds I don't think there is much alternative in systems programming.

closed 2019-09-08 by cirosantilli