cirosantilli/linux-kernel-module-cheat #85: About run gem5 benchmark

guoqinglei, 2019-09-10:
I see that section "18.2. gem5 run benchmark " gives us an very good example on how to run the gem5 benchmark under Buildroot facilities.

If I want to run some other benchmarks or some manually in-house built benchmarks using the Buildroot, how can I do ?
Hopefully, I want to  use Buildroot to do the cross-compiling of some applications and run these application on gem5 using the full system mode.

Anyone can give me some advice ?

cirosantilli, 2019-09-10:
Hi there,

You have to fork this project, and create your own Buildroot packages for your software, and enable / create packages for any dependencies, which is easy usually easy to do (unless you have very complex dependencies).

This is explained at: https://cirosantilli.com/linux-kernel-module-cheat/#add-new-buildroot-packages  @6aa375df2ad3ce2e7d741d09b378503c25547df1

guoqinglei, 2019-09-11:
The possible reason is that I have copied ./config.py file into ./data/config.py without any modifications, then the extended error shows as follows:

tom@tom-pc:~/git/linux-kernel-module-cheat$ ./build-buildroot --arch aarch64 --config 'BR2_PACKAGE_DHRYSTONE=y'
Traceback (most recent call last):
  File "./build-buildroot", line 181, in <module>
    Main().cli()
  File "/home/tom/git/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/tom/git/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/tom/git/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/tom/git/linux-kernel-module-cheat/cli_function.py", line 202, in _get_args
    raise Exception('Unknown key in config file: ' + key + ' ' + config_file) ## added by lgq
Exception: Unknown key in config file: arch /home/tom/git/linux-kernel-module-cheat/data/config.py

Then my question is: if I want to set the default config of build-root by creating file ./data/config.py, how to modify this ./data/config.py file to avoid above errors ?

cirosantilli, 2019-09-11:
Thanks for noticing this, the examaple was wrong, it should be `archs`, fixed at: 3d95fb5f9c904f3cfddcfee1321b3031364765d5

guoqinglei, 2019-09-12:
> Thanks for noticing this, the examaple was wrong, it should be `archs`, fixed at: [3d95fb5](https://github.com/cirosantilli/linux-kernel-module-cheat/commit/3d95fb5f9c904f3cfddcfee1321b3031364765d5)

It works now after i have updated the master.

closed 2019-09-12 by guoqinglei