cirosantilli/linux-kernel-module-cheat #86: About run build-root sample package

guoqinglei, 2019-09-12:
Hi, 

 I refered section "19.5. Add new Buildroot packages" to try to run the sample buildroot package.
But the result is not as expected.

The build command is ./build-buildroot --config 'BR2_PACKAGE_SAMPLE_PACKAGE=y', and the log is as follows:

`tom@tom-pc:~/git/linux-kernel-module-cheat$ ./build-buildroot --config 'BR2_PACKAGE_SAMPLE_PACKAGE=y'
+ cd /home/tom/git/linux-kernel-module-cheat/submodules/buildroot && \
  make \
  O=/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64 \
  BR2_EXTERNAL=../../buildroot_packages/kernel_modules:../../buildroot_packages/lkmc_many_files:../../buildroot_packages/parsec_benchmark:../../buildroot_packages/sample_package \
  qemu_aarch64_virt_defconfig \
;
  GEN     /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/Makefile
#
# configuration written to /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config
#
+ cat \
  /home/tom/git/linux-kernel-module-cheat/buildroot_config/default \
  '>>' \
  /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config \
;
+ cat << 'EOF' >> /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config
BR2_PACKAGE_SAMPLE_PACKAGE=y
BR2_JLEVEL=4
BR2_DL_DIR="/home/tom/git/linux-kernel-module-cheat/out/buildroot/download"
# BR2_LINUX_KERNEL is not set
BR2_GLOBAL_PATCH_DIR="../../patches/global"
BR2_PACKAGE_BUSYBOX_CONFIG_FRAGMENT_FILES="../../busybox_config_fragment"
BR2_PACKAGE_OVERRIDE_FILE="../../buildroot_override"
BR2_ROOTFS_POST_BUILD_SCRIPT="../../rootfs-post-build-script"
BR2_ROOTFS_USERS_TABLES="../../user_table"
BR2_ROOTFS_OVERLAY="../../out/rootfs_overlay/aarch64"
EOF
+ cd /home/tom/git/linux-kernel-module-cheat/submodules/buildroot && \
  make \
  O=/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64 \
  olddefconfig \
;
  GEN     /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/Makefile
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3931:warning: override: reassigning to symbol BR2_CCACHE
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3934:warning: override: reassigning to symbol BR2_ENABLE_DEBUG
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3935:warning: override: reassigning to symbol BR2_GCC_ENABLE_GRAPHITE
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3936:warning: override: reassigning to symbol BR2_GCC_ENABLE_LTO
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3937:warning: override: reassigning to symbol BR2_GCC_ENABLE_OPENMP
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3938:warning: override: reassigning to symbol BR2_OPTIMIZE_0
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3938:warning: override: BR2_OPTIMIZE_0 changes choice state
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3939:warning: override: reassigning to symbol BR2_PTHREAD_DEBUG
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3940:warning: override: reassigning to symbol BR2_TOOLCHAIN_BUILDROOT_CXX
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3941:warning: override: reassigning to symbol BR2_TOOLCHAIN_BUILDROOT_FORTRAN
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3943:warning: override: reassigning to symbol BR2_TOOLCHAIN_BUILDROOT_GLIBC
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3943:warning: override: BR2_TOOLCHAIN_BUILDROOT_GLIBC changes choice state
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3944:warning: override: reassigning to symbol BR2_TOOLCHAIN_BUILDROOT_WCHAR
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3947:warning: override: reassigning to symbol BR2_TARGET_ROOTFS_CPIO
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3948:warning: override: reassigning to symbol BR2_TARGET_ROOTFS_EXT2
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3949:warning: override: reassigning to symbol BR2_TARGET_ROOTFS_EXT2_SIZE
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3950:warning: override: reassigning to symbol BR2_TARGET_ROOTFS_SQUASHFS
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3951:warning: override: reassigning to symbol BR2_TARGET_ROOTFS_INITRAMFS
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3953:warning: override: reassigning to symbol BR2_TARGET_ROOTFS_TAR
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3956:warning: override: reassigning to symbol BR2_PACKAGE_HOST_GDB
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3962:warning: override: reassigning to symbol BR2_PACKAGE_STRACE
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3965:warning: override: reassigning to symbol BR2_PACKAGE_HOST_DTC
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3972:warning: override: reassigning to symbol BR2_PACKAGE_SAMPLE_PACKAGE
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3973:warning: override: reassigning to symbol BR2_JLEVEL
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3974:warning: override: reassigning to symbol BR2_DL_DIR
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3975:warning: override: reassigning to symbol BR2_LINUX_KERNEL
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3976:warning: override: reassigning to symbol BR2_GLOBAL_PATCH_DIR
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3977:warning: override: reassigning to symbol BR2_PACKAGE_BUSYBOX_CONFIG_FRAGMENT_FILES
/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/.config:3978:warning: override: reassigning to symbol BR2_PACKAGE_OVERRIDE_FILE
[... 629 lines cut ...]
mkdir -p /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/buildroot-fs/ext2
echo '#!/bin/sh' > /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/buildroot-fs/ext2/fakeroot
echo "set -e" >> /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/buildroot-fs/ext2/fakeroot
printf '   	mkdir -p /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/buildroot-fs/ext2/target\n	tar xf /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/buildroot-fs/rootfs.common.tar -C /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/buildroot-fs/ext2/target\n' >> /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/buildroot-fs/ext2/fakeroot
printf '   \n' >> /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/buildroot-fs/ext2/fakeroot
printf '   	rm -f /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/images/rootfs.ext2\n	/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/host/sbin/mkfs.ext4 -d /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/buildroot-fs/ext2/target -r 1 -N 0 -m 5 -L "" -O ^64bit /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/images/rootfs.ext2 "512M" || { ret=$?; echo "*** Maybe you need to increase the filesystem size (BR2_TARGET_ROOTFS_EXT2_SIZE)" 1>&2; exit $ret; }\n' >> /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/buildroot-fs/ext2/fakeroot
chmod a+x /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/build/buildroot-fs/ext2/fakeroot
PATH="/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/host/bin:/home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" /home/tom/git/linux-kernel- [... cut ...]
Creating filesystem with 131072 4k blocks and 32768 inodes
Filesystem UUID: 56ce8948-41a5-4cbd-95dd-8e62165a8328
Superblock backups stored on blocks: 
	32768, 98304

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (4096 blocks): done
Copying files into the device: done
Writing superblocks and filesystem accounting information: done

ln -sf rootfs.ext2 /home/tom/git/linux-kernel-module-cheat/out/buildroot/build/default/aarch64/images/rootfs.ext4
time 00:00:13
`

guoqinglei, 2019-09-14:
I have run "./gem5-shell" and I have seen the following log:

```
tom@tom-pc:~/git/linux-kernel-module-cheat$ ./gem5-shell 
+ /home/tom/git/linux-kernel-module-cheat/out/gem5/default/m5term localhost 3456
==== m5 slave terminal: Terminal 0 ====
<6>[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fc0f0]
<5>[    0.000000] Linux version 5.2.1-g527a3db36 (lkmc@527a3db363a3bd7e6ae0a77da809e01847a9931c) (gcc version 7.3.0 (Buildroot 2018.08-g653eaa17)) #1 SMP Thu Jan 1 00:00:00 UTC 1970
<6>[    0.000000] Machine model: V2P-CA15
<5>[    0.000000] Memory limited to 256MB
<6>[    0.000000] efi: Getting EFI parameters from FDT:
<6>[    0.000000] efi: UEFI not found.
<7>[    0.000000] On node 0 totalpages: 65536
<7>[    0.000000]   DMA32 zone: 1024 pages used for memmap
<7>[    0.000000]   DMA32 zone: 0 pages reserved
<7>[    0.000000]   DMA32 zone: 65536 pages, LIFO batch:15
<6>[    0.000000] percpu: Embedded 30 pages/cpu s83928 r8192 d30760 u122880
<7>[    0.000000] pcpu-alloc: s83928 r8192 d30760 u122880 alloc=30*4096
<7>[    0.000000] pcpu-alloc: [0] 0 
<6>[    0.000000] Detected PIPT I-cache on CPU0
<6>[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
<4>[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
<6>[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64512
<5>[    0.000000] Kernel command line: earlyprintk=pl011,0x1c090000 lpj=19988480 rw loglevel=8 mem=256MB root=/dev/sda console_msg_format=syslog nokaslr norandmaps panic=-1 printk.devkmsg=on printk.time=y rw console=ttyAMA0 - lkmc_home=/lkmc lkmc_eval_base64="Li9zYW1wbGVfcGFja2FnZS5vdXQ="
<6>[    0.000000] Dentry cache hash table entries: 32768 (order: 6, 262144 bytes)
<6>[    0.000000] Inode-cache hash table entries: 16384 (order: 5, 131072 bytes)
<6>[    0.000000] Memory: 232648K/262144K available (6652K kernel code, 828K rwdata, 2180K rodata, 1088K init, 661K bss, 29496K reserved, 0K cma-reserved)
<6>[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
<6>[    0.000000] ftrace: allocating 21681 entries in 85 pages
<6>[    0.000000] rcu: Hierarchical RCU implementation.
<6>[    0.000000] rcu: 	RCU event tracing is enabled.
<6>[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=1.
<6>[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
<6>[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=1
<6>[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
<5>[    0.000000] random: get_random_bytes called from start_kernel+0x32c/0x4d0 with crng_init=0
<6>[    0.000001] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
<6>[    0.000054] arch_timer: cp15 timer(s) running at 25.16MHz (virt).
<6>[    0.000058] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x5cdd39714, max_idle_ns: 440795202620 ns
<6>[    0.000061] sched_clock: 56 bits at 25MHz, resolution 39ns, wraps every 4398046511084ns
<6>[    0.000102] Console: colour dummy device 80x25
<6>[    0.000110] Calibrating delay loop (skipped) preset value.. 9994.24 BogoMIPS (lpj=19988480)
<6>[    0.000116] pid_max: default: 32768 minimum: 301
<6>[    0.000207] Mount-cache hash table entries: 512 (order: 0, 4096 bytes)
<6>[    0.000213] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes)
<6>[    0.000612] ASID allocator initialised with 128 entries
<6>[    0.000656] rcu: Hierarchical SRCU implementation.
<6>[    0.000866] EFI services will not be available.
<6>[    0.000907] smp: Bringing up secondary CPUs ...
<6>[    0.000912] smp: Brought up 1 node, 1 CPU
<6>[    0.000916] SMP: Total of 1 processors activated.
<6>[    0.000920] CPU features: detected: 32-bit EL0 Support
<6>[    0.000925] CPU features: detected: Scalable Vector Extension
<6>[    0.000929] CPU features: detected: Data cache clean to the PoU not required for I/D coherence
<6>[    0.001355] SVE: maximum available vector length 16 bytes per vector
<6>[    0.001360] SVE: default vector length 16 bytes per vector
<6>[    0.001365] CPU: All CPU(s) started at EL1
<6>[    0.001373] alternatives: patching kernel code
<6>[    0.001672] devtmpfs: initialized
<6>[    0.002344] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
<6>[    0.002351] futex hash table entries: 256 (order: 2, 16384 bytes)
<6>[    0.002773] DMI not present or invalid.
<6>[    0.002843] NET: Registered protocol family 16
<6>[    0.080454] hw-breakpoint: found 2 breakpoint and 2 watchpoint registers.
<6>[    0.080478] DMA: preallocated 256 KiB pool for atomic allocations
<6>[    0.080483] Serial: AMBA PL011 UART driver
<3>[    0.080711] OF: amba_device_add() failed (-2) for /rtc@1c170000
<6>[    0.080865] 1c090000.uart: ttyAMA0 at MMIO 0x1c090000 (irq = 7, base_baud = 0) is a PL011 rev3
<6>[    0.081153] printk: console [ttyAMA0] enabled
<3>[    0.081661] OF: amba_device_add() failed (-2) for /kmi@1c070000
[... 63 lines cut ...]
<5>[    0.292921] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
<5>[    0.293540] sd 0:0:0:0: [sda] Attached SCSI disk
<6>[    0.294263] EXT4-fs (sda): mounted filesystem with ordered data mode. Opts: (null)
<6>[    0.294276] VFS: Mounted root (ext4 filesystem) on device 8:0.
<6>[    0.294329] devtmpfs: mounted
<6>[    0.295533] Freeing unused kernel memory: 1088K
<6>[    0.304388] Run /sbin/init as init process
[... cut ...]
<3>[    0.310572] 9pnet_virtio: no channels available for device host_out_rootfs_overlay
mount: mounting host_out_rootfs_overlay on /mnt/9p/out_rootfs_overlay failed: No such file or directory
<3>[    0.310977] overlayfs: overlapping upperdir path
mount: mounting overlay on /mnt/overlay failed: Too many levels of symbolic links
hello S98
/etc/init.d/S98: eval: line 1: ./sample_package.out: not found
hello .profile
/lkmc

```

I still can not see the expected output "hello sample_package" here, why ?

cirosantilli, 2019-09-14:
OK, first there was a documentation bug which I fixed at: e14f0e367952dad3f4e076caac1f5d12404d91b7 should be instead:

```
./run --eval-after '/sample_package.out'
```

because the file is placed in the root currently.

If that doesn't work, try to do a `find` in `linux-kernel-module-cheat-out/buildroot/build/default/x86_64` to see if it was built and installed into `target/`, and if not also try to force a reubuild with:

```
./build-buildroot --config 'BR2_PACKAGE_SAMPLE_PACKAGE=y' -- sample_package-reconfigure
```

guoqinglei, 2019-09-14:
> OK, first there was a documentation bug which I fixed at: [e14f0e3](https://github.com/cirosantilli/linux-kernel-module-cheat/commit/e14f0e367952dad3f4e076caac1f5d12404d91b7) should be instead:
> 
> ```
> ./run --eval-after '/sample_package.out'
> ```
> 
> because the file is placed in the root currently.
> 
> If that doesn't work, try to do a `find` in `linux-kernel-module-cheat-out/buildroot/build/default/x86_64` to see if it was built and installed into `target/`, and if not also try to force a reubuild with:
> 
> ```
> ./build-buildroot --config 'BR2_PACKAGE_SAMPLE_PACKAGE=y' -- sample_package-reconfigure
> ```

I have rebuild with sample package, and found the sample_package.out in the targe/ direcotry.
Now I run with : 
`./run --eval-after '/sample_package.out'`
And it works.

closed 2019-09-17 by guoqinglei