cirosantilli/linux-kernel-module-cheat #98: arch linux - fail to build, configure fails with -Werror

YanB25, 2019-11-03:
When I run `./build --download-dependencies qemu-buildroot --no-apt` following section 1.1.1. QEMU Buildroot setup getting started, an error occurs.

Below are the logs.
```
make -C '/home/yanbin/linux-kernel-module-cheat/out/linux/default/x86_64' M='/home/yanbin/linux-kernel-module-cheat/out/kernel_modules/x86_64/kernel_modules'
make[1]: Entering directory '/home/yanbin/linux-kernel-module-cheat/out/linux/default/x86_64'
  Building modules, stage 2.
  MODPOST 47 modules
make[1]: Leaving directory '/home/yanbin/linux-kernel-module-cheat/out/linux/default/x86_64'
+ ./copy-overlay \
  --no-show-time \
;
+ cp \
  /home/yanbin/linux-kernel-module-cheat/rootfs_overlay/etc/ld.so.cache \
  /home/yanbin/linux-kernel-module-cheat/out/rootfs_overlay/x86_64/etc/ld.so.cache \
;
+ ./build-qemu \
  --no-show-time \
;
+ cd /home/yanbin/linux-kernel-module-cheat/out/qemu/default/opt && \
  PATH="/usr/lib/ccache:${PATH}" \
  /home/yanbin/linux-kernel-module-cheat/submodules/qemu/configure \
  --enable-trace-backends=simple \
  --target-list=x86_64-softmmu \
  --enable-sdl \
;

ERROR: configure test passed without -Werror but failed with -Werror.
       This is probably a bug in the configure script. The failing command
       will be at the bottom of config.log.
       You can run configure with --disable-werror to bypass this check.

Traceback (most recent call last):
  File "./build", line 656, in <module>
    Main().cli()
  File "/home/yanbin/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/yanbin/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/yanbin/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/yanbin/linux-kernel-module-cheat/common.py", line 1322, in main
    ret = self.timed_main()
  File "./build", line 653, in timed_main
    component.build(self.env['arch'])
  File "./build", line 57, in build
    self.build_callback()
  File "./build", line 447, in f
    lkmc.import_path.import_path_main(component_file)(**args)
  File "/home/yanbin/linux-kernel-module-cheat/common.py", line 650, in __call__
    return super().__call__(**kwargs)
  File "/home/yanbin/linux-kernel-module-cheat/cli_function.py", line 149, in __call__
   return self._do_main(kwargs)
  File "/home/yanbin/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/yanbin/linux-kernel-module-cheat/common.py", line 1322, in main
    ret = self.timed_main()
  File "/home/yanbin/linux-kernel-module-cheat/common.py", line 1745, in timed_main
    return self.build()
  File "/home/yanbin/linux-kernel-module-cheat/build-qemu", line 44, in build
    cwd=build_dir
  File "/home/yanbin/linux-kernel-module-cheat/shell_helpers.py", line 416, in run_cmd
    raise e
Exception: Command exited with status: 1
```

I wished to run configure with `--disable-werror`, but I failed to figure out where I should change the codes.

## some infos
I have run `pacman -Syu` and reboot.

```
$ uname -a
Linux arch-linux 5.3.8-arch1-1 #1 SMP PREEMPT @1572357769 x86_64 GNU/Linux
```

```
$ pkg-config --version
1.6.3
```

cirosantilli, 2019-11-03:
Hi Benson,

From the logs, QEMU failed to configure. Can you have a look at the `config.log.` mentioned in the error message and share / Google its error message?

YanB25, 2019-11-04:
Hi Cirosantilli,

The problem in qemu has been fixed by [this patch](https://lists.sr.ht/~philmd/qemu/%3C20190617114114.24897-1-berrange%40redhat.com%3E). 

However, this patch is not merged into `submodules/qemu`. Moreover, it seems that when I run `./build --download-dependencies qemu-buildroot --no-apt`, `submodules/qemu` will automatically "checkout" to the version forked in this repo(where this patch is not merged).

After manually apply this patch to the submodules/qemu source codes, I encounter a lot more error like this one(the error has been fixed, but the patch has not been merged).

cirosantilli, 2019-11-04:
Awesome, thanks for looking into this!

For now, just cherry pick patches / commits into your tree, and if you really need a reproducible setup, fork this repo and make the submodule point to your own QEMU fork with the patches.

Once all we notice that all fixing patches are merged, let me know and I will update the QEMU submodule in master.

YanB25, 2019-11-06:
There's still problem building this repo on arch linux. I fail to build even all the submodules are updated to the latest commit. Maybe I'll try to dig deeper later but for now I give up and try it on Ubuntu.

cirosantilli, 2019-11-06:
OK, let me know the failure message so others can Google it as well.

cirosantilli, 2019-12-16:
QEMU v4.2.0 builds fine on Ubuntu 19.20 GCC 9.2 BTW.