cirosantilli/linux-kernel-module-cheat #104: SMT support

Eleanor99, 2019-11-11:
I am new to gem5.
Can I simulate simultaneous multithreading CPUs with this? Does gem5 support smt? I've tried looking into gem5 forums, but could not get an answer.

cirosantilli, 2019-11-11:
Hi Eleanor,

What is SMT? :-)

cirosantilli, 2019-11-11:
OK, if you mean https://en.wikipedia.org/wiki/Simultaneous_multithreading in a single sycall emulation CPU, I believe that the answer is no, I think gem5 only uses unused CPUs on fork and frees them up on join.

You would either need to use syscall emulation, or patch gem5, or run one thread per CPU core if that is acceptable.

cirosantilli, 2019-11-12:
Wait, actually,  `se.py` has a `--smt` options,  I didn't know that. And the syntax appears to be:

```
--smt
-c 'tests/test-progs/hello/bin/x86/linux/hello;tests/test-
progs/hello/bin/x86/linux/hello'
```

as per: https://gem5-users.gem5.narkive.com/DCqFpzOR/gem5-smt-in-se-mode and help says:

```
  --smt                                        Only used if multiple programs
                        are specified. If true,                       then the
                        number of threads per cpu is same as the
                        number of programs.
```

Have you tried that, and how does it fail?

Eleanor99, 2020-01-24:
I didn't realize you actually replied. 

I tried `--smt` but this is what I get:
`
gem5 Simulator System.  http://gem5.org
gem5 is copyrighted software; use the --copyright option for details.

gem5 compiled Jan 20 2020 22:23:14
gem5 started Jan 24 2020 22:48:21
gem5 executing on nirmal-cadsl2, pid 5776
command line: ./build/X86/gem5.opt ./configs/example/se.py --cpu-type=DerivO3CPU --caches -c 'tests/test-progs/hello/bin/x86/linux/hello;tests/test-progs/hello/bin/x86/linux/hello' --smt

Global frequency set at 1000000000000 ticks per second
warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (512 Mbytes)
0: system.remote_gdb: listening for remote gdb on port 7000
0: system.remote_gdb: listening for remote gdb on port 7001
panic: panic condition !intMasterPort.isConnected() occurred: Int port not connected to anything!
Memory Usage: 669696 KBytes
Program aborted at tick 0
--- BEGIN LIBC BACKTRACE ---
./build/X86/gem5.opt(_Z15print_backtracev+0x2c)[0x5635f509141c]
./build/X86/gem5.opt(_Z12abortHandleri+0x4a)[0x5635f50a376a]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x12890)[0x7fda9ac38890]
/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7)[0x7fda9989fe97]
/lib/x86_64-linux-gnu/libc.so.6(abort+0x141)[0x7fda998a1801]
./build/X86/gem5.opt(+0x52147f)[0x5635f4e1447f]
./build/X86/gem5.opt(_ZN6X86ISA10Interrupts4initEv+0x117)[0x5635f5175f67]
./build/X86/gem5.opt(+0x1190136)[0x5635f5a83136]
./build/X86/gem5.opt(+0x6fcbe4)[0x5635f4fefbe4]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6ff3)[0x7fda9aee6753]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7d8)[0x7fda9b024908]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x5bf6)[0x7fda9aee5356]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7d8)[0x7fda9b024908]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x5bf6)[0x7fda9aee5356]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7d8)[0x7fda9b024908]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCode+0x19)[0x7fda9aedf5c9]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6ac0)[0x7fda9aee6220]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7d8)[0x7fda9b024908]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x5bf6)[0x7fda9aee5356]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x7d8)[0x7fda9b024908]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCode+0x19)[0x7fda9aedf5c9]
/usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyRun_StringFlags+0x76)[0x7fda9af8f6e6]
./build/X86/gem5.opt(_Z6m5MainiPPc+0x63)[0x5635f50a21a3]
./build/X86/gem5.opt(main+0x38)[0x5635f4dd06b8]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7)[0x7fda99882b97]
./build/X86/gem5.opt(_start+0x2a)[0x5635f4e015ba]
--- END LIBC BACKTRACE ---
`
Seems like there is a work around for this:
https://www.mail-archive.com/gem5-users@gem5.org/msg04271.html
http://reviews.gem5.org/r/1281/#review2999
This patch can be applied on an older version of gem5 on ubuntu 12.04.

However I want to run a multithreaded program. It doesn't look like gem5 supports it. Like you said, the number of threads can only be equal to number of programs.

As for Full System, This is what I found in the file  [https://github.com/gem5/gem5/blob/377898c4034c72b84b2662ed252fa25079a4ea62/src/cpu/o3/cpu.cc#L313](src/cpu/o3/cpu.cc) :

> // SMT is not supported in FS mode yet.

SaiVK, 2020-05-15:
Was this issue w.r.t SMT on Gem5 resolved ?

cirosantilli, 2020-05-15:
@SaiVK nope, nothing as far as I know. But I have the feeling it would be easy to get it working again, the current error message are likely simple stuff that rotted.

SimonLammer, 2023-07-21:
@Eleanor99 Have you managed to simulate SMT in gem5 by now?

ssari-memory, 2023-08-06:
Is there an update for this issue? Since `se` support will be deprecated, I guess there will be no support for SMT on `se` either? With the applied patch, there is still errors with `--smt `option. Should anyone who wants to simulate SMT processors, will have to use `fs` in this case?

cirosantilli, 2023-08-06:
> Since `se` support will be deprecated

Do you have a source for this?

ssari-memory, 2023-08-07:
> > Since `se` support will be deprecated
> 
> @ssari-memory Do you have a source for this?

[source](https://github.com/gem5/gem5/blob/48b4788bbe7a091b917aed51236bca2b0456cf37/configs/example/se.py#L30)

cirosantilli, 2023-08-07:
Thanks. But it is just the script, you can just copy it out of tree, underlying functionality will still be there.

I kept telling them, fs.py and se.py are important examples, but that project simply cannot understand that people want to easily and simply run basic stuff before making their own complex scripts. What can I do.