cirosantilli/linux-kernel-module-cheat #105: Rebuilt busybox::wget changes not showing.

sarnobat, 2019-11-12:
(further to our conversation here: https://stackoverflow.com/questions/47557262/how-to-download-the-torvalds-linux-kernel-master-recompile-it-and-boot-it-wi/49349237?noredirect=1#comment103838604_49349237)

The entire `./build` succeeds after I add a simple hello world `printf` statement in `output/build/busybox-1.29.3/networking/wget.c::wget_main()`  (unlike when I was trying to do the same on the non-cheat version of buildroot which gives a segmentation fault).

However, when I run `wget` inside the VM I don't see any of my `printf` output (or modifications to the wget `usage.h` entry that I try).

I'm guessing that I don't understand linking or rebuilding properly. I thought it would be as simple as `make busybox-rebuild linux-rebuild` but that doesn't seem to be enough.

Can you help me get my changes to busybox's `wget.c` to be built and reflected in my running VM (in qemu)?

I really appreciate your help so far, and this repo is so valuable.

cirosantilli, 2019-11-12:
Hi sarnobat,

I think Buildroot uses per package timestamps to quickly decide if something needs to be rebuilt.

So when you want to rebuilt it, you must explicitly request it with the `make PKG-rebuild` target, so in this case I believe `busybox-rebuild`.

From this repo, you can do that with something like (untested):

```
./build-buildroot -- busybox-rebuild
```

`linux-rebuild` would only rebuild the Linux kernel which is in a package called `linux` I believe.

Note however that the cleaner solution is to keep your busybox source out of tree in a git submodule as done for most packages under submodules/, apply all per package Buildroot patches, and tell Buildroot to use it with BR2_EXTERNAL: https://buildroot.org/downloads/manual/manual.html#outside-br-custom

cirosantilli, 2019-11-13:
Said wrong, OVERRIDE_SRCDIR, not BR2_EXTERNAL.

sarnobat, 2019-11-13:
Hey it works!!
You are some kind of Linux kernel god. I'm sure I'll have followup questions but this one is resolved. Thank you.

cirosantilli, 2019-11-13:
Great! I've just been doing a bit more Buildroot than any sane person should! :-) Closing as resolved.

closed 2019-11-13 by cirosantilli