cirosantilli/linux-kernel-module-cheat #106: Debugging a device driver (intel/e1000e)

sarnobat, 2019-11-23:
I'm having trouble figuring out how to debug a network interface card. I'm trying to follow this: https://github.com/cirosantilli/linux-kernel-module-cheat#gdb-step-debug-kernel-module but gdb still gives 

## Preconditions
Successfully completes (and I can run wget on an external http endpoint):
```ifup -a```

## input
```
(gdb) break e1000e_up
```

## error message
`Make breakpoint pending on future shared library load? (y or [n])` 

## Other info

I also tried this but don't see the functions I expect:

```
(gdb) list functions e1000
```

## device info is:

```
root@buildroot# lsmod

Module                  Size  Used by    Not tainted
/sys/devices/pci0000:00/0000:00:03.0/net/eth0
```

And the hardware info (from the host machine) is:
```
$ lshw -class network

WARNING: you should run this program as super-user.
  *-network
       description: Wireless interface
       product: Wireless 8265 / 8275
       vendor: Intel Corporation
       physical id: 0
       bus info: pci@0000:3a:00.0
       logical name: wlp58s0
       version: 78
       serial: f8:63:3f:2c:78:dd
       width: 64 bits
       clock: 33MHz
       capabilities: bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=iwlwifi driverversion=4.15.0-70-generic firmware=34.0.1 latency=0 link=no multicast=yes wireless=IEEE 802.11
       resources: irq:131 memory:dc100000-dc101fff
  *-network
       description: Ethernet interface
       product: Ethernet Connection (4) I219-V
       vendor: Intel Corporation
       physical id: 1f.6
       bus info: pci@0000:00:1f.6
       logical name: eno1
       version: 21
       serial: f4:4d:30:6d:73:63
       size: 1Gbit/s
       capacity: 1Gbit/s
       width: 32 bits
       clock: 33MHz
       capabilities: bus_master cap_list ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=e1000e driverversion=3.2.6-k duplex=full firmware=0.1-4 ip=192.168.1.5 latency=0 link=yes multicast=yes port=twisted pair speed=1Gbit/s
       resources: irq:129 memory:dc200000-dc21ffff
WARNING: output may be incomplete or inaccurate, you should run this program as super-user.
```

cirosantilli, 2019-11-23:
Hi there,

Are you sure that the kernel is that device?

`out/linux/default/x86_64/.config` contains:

```
# CONFIG_E100 is not set
# CONFIG_E1000 is not set
# CONFIG_E1000E is not set
```

try to enable those.

Current boot has messages:

```
<6>[    2.109387] 8139cp: 8139cp: 10/100 PCI Ethernet driver v1.3 (Mar 22, 2004)                       
<6>[    2.301500] 8139cp 0000:00:03.0 eth0: RTL-8139C+ at 0x(____ptrval____), 52:54:00:12:34:56, IRQ 11
```

which might be the one actually being used, not sure.

Then also ensure that QEMU is exposing that device, e.g. the only net device on `info qtree` appears to be https://cirosantilli.com/linux-kernel-module-cheat/#qemu-monitor

```
dev: rtl8139, id ""        
  mac = "52:54:00:12:34:56"
  netdev = "net0"          
```

so I think `rtl8139` is the one it has. Also I have a comment on `linux_config/default` suggesting that, had forgotten about it:

```
# If given, we can use QEMU 2.9.0 default x86 networking without any -net or -netdev options,
# since E1000 is the default networking device as mentioned at:
# https://en.wikibooks.org/w/index.php?title=QEMU/Networking&oldid=3268753
#CONFIG_E1000=y
```

It would also be good to learn how to list all available net devices / select them with QEMU CLI. `man qemu-system-x75_64` has a related example:

```
-netdev tap,id=nd0,ifname=tap0 -device e1000,netdev=nd0 \
-netdev tap,id=nd1,ifname=tap1 -device rtl8139,netdev=nd1
```

sarnobat, 2019-11-23:
Thanks for the quick response as usual. I'll look into it soon and reopen the ticket if I need further help.

closed 2019-11-23 by sarnobat

sarnobat, 2019-11-26:
Just confirming, it does find this breakpoint and stop there:

```
#0  cp_start_xmit (skb=0xffff88800e134600, dev=0xffff88800f3d0000)
    at buildroot-arm.2019.02.6.cheat/linux-kernel-module-cheat/submodules/linux/drivers/net/ethernet/realtek/8139cp.c:733
```