cirosantilli/linux-kernel-module-cheat #108: Buildroot with GCC 9.2 build fails with "xgcc: No such file or directory"

cirosantilli, 2019-12-16:
At https://github.com/cirosantilli/linux-kernel-module-cheat/tree/buildroot-2019.11-1 which has buildroot 2019.11 and GCC 9.2:

```
  cc1-checksum.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a  -L/mnt/sda3/linux-kernel-module-cheat-out/buildroot/build/default/x86_64/host/lib -lisl -L/mnt/sda3/linux-kernel-module-cheat-out/buildroot/build/default/x86_64/host/lib -L/mnt/sda3/linux-kernel-module-cheat-out/buildroot/build/default/x86_64/host/lib -L/mnt/sda3/linux-kernel-module-cheat-out/buildroot/build/default/x86_64/host/lib -lmpc -lmpfr -lgmp -rdynamic -ldl  -L./../zlib -lz
/mnt/sda3/linux-kernel-module-cheat-out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/./gcc/xgcc -B/mnt/sda3/linux-kernel-module-cheat-out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/./gcc/ -xc++ -nostdinc /dev/null -S -o /dev/null -fself-test=../../gcc/testsuite/selftests
/bin/bash: /mnt/sda3/linux-kernel-module-cheat-out/buildroot/build/default/x86_64/build/host-gcc-final-custom/build/./gcc/xgcc: No such file or directory
make[4]: *** [../../gcc/cp/Make-lang.in:178: s-selftest-c++] Error 127
make[4]: *** Waiting for unfinished jobs....
../../gcc/gcc.c: In function ‘long unsigned int get_random_number()’:
../../gcc/gcc.c:9639:12: warning: ignoring return value of ‘ssize_t read(int, void*, size_t)’, declared with attribute warn_unused_result [-Wunused-result]
 9639 |       read (fd, &ret, sizeof (HOST_WIDE_INT));
      |       ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../gcc/gcc.c: In function ‘void do_report_bug(const char**, int, char**, char**)’:
../../gcc/gcc.c:6999:9: warning: ignoring return value of ‘ssize_t write(int, const void*, size_t)’, declared with attribute warn_unused_result [-Wunused-result]
 6999 |   write (fd, "\n//", 3);
      |   ~~~~~~^~~~~~~~~~~~~~~
../../gcc/gcc.c:7002:13: warning: ignoring return value of ‘ssize_t write(int, const void*, size_t)’, declared with attribute warn_unused_result [-Wunused-result]
 7002 |       write (fd, " ", 1);
      |       ~~~~~~^~~~~~~~~~~~
../../gcc/gcc.c:7003:13: warning: ignoring return value of ‘ssize_t write(int, const void*, size_t)’, declared with attribute warn_unused_result [-Wunused-result]
 7003 |       write (fd, new_argv[i], strlen (new_argv[i]));
      |       ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../gcc/gcc.c:7005:9: warning: ignoring return value of ‘ssize_t write(int, const void*, size_t)’, declared with attribute warn_unused_result [-Wunused-result]
 7005 |   write (fd, "\n\n", 2);
      |   ~~~~~~^~~~~~~~~~~~~~~
rm gcov.pod gfdl.pod gpl.pod cpp.pod gcc.pod gcov-dump.pod gfortran.pod gcov-tool.pod fsf-funding.pod
make[3]: *** [Makefile:4322: all-gcc] Error 2
make[2]: *** [Makefile:955: all] Error 2
make[1]: *** [package/pkg-generic.mk:241: /mnt/sda3/linux-kernel-module-cheat-out/buildroot/build/default/x86_64/build/host-gcc-final-custom/.stamp_built] Error 2
make: *** [Makefile:84: _all] Error 2
Traceback (most recent call last):
  File "./build", line 656, in <module>
    Main().cli()
  File "/home/ciro/bak/git/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/ciro/bak/git/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/ciro/bak/git/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/ciro/bak/git/linux-kernel-module-cheat/common.py", line 1372, in main
    ret = self.timed_main()
  File "./build", line 653, in timed_main
    component.build(self.env['arch'])
  File "./build", line 57, in build
    self.build_callback()
  File "./build", line 447, in f
    lkmc.import_path.import_path_main(component_file)(**args)
  File "/home/ciro/bak/git/linux-kernel-module-cheat/common.py", line 685, in __call__
    return super().__call__(**kwargs)
  File "/home/ciro/bak/git/linux-kernel-module-cheat/cli_function.py", line 149, in __call__
    return self._do_main(kwargs)
  File "/home/ciro/bak/git/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/ciro/bak/git/linux-kernel-module-cheat/common.py", line 1372, in main
    ret = self.timed_main()
  File "/home/ciro/bak/git/linux-kernel-module-cheat/common.py", line 1795, in timed_main
    return self.build()
  File "/home/ciro/bak/git/linux-kernel-module-cheat/build-buildroot", line 166, in build
    cwd=self.env['buildroot_source_dir'],
  File "/home/ciro/bak/git/linux-kernel-module-cheat/shell_helpers.py", line 432, in run_cmd
    raise e
Exception: Command exited with status: 2
```

RaspberryPiFan, 2020-05-05:
I'm also having this issue with buildroot 2020.02 (not from your repo). I could fix it by deleting the out/build/<gcc with issue> directory and running make again. Had to do this evry time I build, so I used binutils 2.32, I assume some build variables are set wrong at first build, so when I started build again after deleting the dir, some variables other builds set aren't set anymore.

fidomax, 2020-05-06:
Just apply this patch to gcc:
--- a/gcc/Makefile.in	2019-09-04 10:05:38.483715500 +0300
+++ b/gcc/Makefile.in	2019-04-25 17:32:16.000000000 +0300
@@ -1961,7 +1961,7 @@
 # require the selftests to be run by defining their selftest-<LANG> as
 # s-selftest-<LANG>.  Otherwise, they should define it as empty.
 
-SELFTEST_TARGETS = @selftest_languages@
+SELFTEST_TARGETS = 
 selftest: $(SELFTEST_TARGETS)
 
 # Recompile all the language-independent object files.

MilhouseVH, 2020-06-02:
LibreELEC has been seeing this issue with gcc-9.x for a while - can't be certain when it started, but we didn't have this issue with gcc-8.3.0 so it definitely appeared with either 9.1 or 9.2, and continues with 9.3.

I've tested the patch in this PR (see [here](https://github.com/LibreELEC/LibreELEC.tv/compare/master...MilhouseVH:le10_stuff)) and unfortunately I had another [failure](http://milhouse.libreelec.tv/other/gcc-with-pr108.log.txt) when clean building `gcc-9.3.0` today, so I don't think the patch is fixing the issue (either that, or LibreELEC has a different but very similar issue). These failures are extremely random (maybe one failure in 20) and difficult to reproduce on demand (it only ever fails when you don't want it to!) so it's entirely possible the patch may only have made it appear fixed, or maybe makes it harder to reproduce.

LibreELEC is building with `make-4.3` and `ccache-3.7.7` in case that makes a difference. Is there any upstream mailing list discussion, or bug?

MilhouseVH, 2020-06-02:
Oh... oh FFS. Just took another look at the patch (in the LE repo) and realised it's not being applied correctly (it's patching the wrong path!) so ignore the above comment - I'll run some more clean build tests, this time with the patch being applied correctly!!! :argh: