cirosantilli/linux-kernel-module-cheat #110: QEMU x86_64 -nographic full system breaks host Bash terminal line wrapping after simulation ends

cirosantilli, 2019-12-24:
At LKMC 90f3701f834aadb75f6d92fd8157fe0cbab0fc52 QEMU hacked to 4.2.0 (also happened before) Ubuntu 19.10, open a "fresh" new GNOME terminal running bash from Dash, and then just:

```
./run
```

and quit the simulation however you want, e.g. `Ctrl A-X`.

Now, copy paste a long character sequence multiple times on the host terminal QEMU just ran on, and after the right end of screen is reached, the pasted text stops wrapping and just disappears beyond the terminal.

Sometimes I've also seen it wrap back to the left, but on the same line as the current one, overwriting whatever is there, or other crazier effects.

What triggers me the most is that I don't know how to examine the shell / terminal state to find exactly the name of whatever property it is that controls this. I've tried a gazillion things like:

```
shopt checkwinsize
```

but this just beyond my Google fu, bibliography:

- https://unix.stackexchange.com/questions/105958/terminal-prompt-not-wrapping-correctly

rico-chet, 2021-05-18:
This is really a nasty behavior. I worked around it by running this `expect` script instead of starting QEMU directly:

````
#!/usr/bin/expect -f

set timeout -1
log_user 0; # suppress the terminal damaging bytes
spawn qemu-system-i386 \
        -M pc \
        -kernel output/images/bzImage \
        -drive file=output/images/rootfs.ext2,if=virtio,format=raw \
        -append {rootwait root=/dev/vda console=ttyS0} \
        -net nic,model=virtio \
        -net user \
        -nographic \

expect -re "(Linux version .*)"
puts $expect_out(1,string); # version information is important, don't suppress
log_user 1; # get back to normal

interact
````

turistu, 2023-04-06:
If you want just to get rid of that (and not so much about exploring the mysteries of terminal escapes), you can apply this patch to [seabios](https://gitlab.com/qemu-project/seabios) rebuild it, and then let qemu use it instead of the default `pc-bios/bios-256k.bin`.
```
diff --git a/src/sercon.c b/src/sercon.c
index 3019d9b..3e8fba0 100644
--- a/src/sercon.c
+++ b/src/sercon.c
@@ -346,10 +346,12 @@ static void sercon_1000(struct bregs *regs)
     SET_LOW(sercon_row_last, 0);
     SET_LOW(sercon_attr_last, 0);

+    /*
     sercon_term_reset();
     sercon_term_no_linewrap();
     if (clearscreen)
         sercon_term_clear_screen();
+       */
 }

 /* Set text-mode cursor shape */
```