cirosantilli/linux-kernel-module-cheat #118: design-goals and supported-hosts links don't go anywhere

mbigras, 2020-03-22:
Hi @cirosantilli!

I was reading through the readme in these sections:

https://github.com/cirosantilli/linux-kernel-module-cheat#1-getting-started
https://github.com/cirosantilli/linux-kernel-module-cheat#111-qemu-buildroot-setup-getting-started

And there are two links in the READM.adoc that don't go anywhere

https://github.com/cirosantilli/linux-kernel-module-cheat#design-goals
https://github.com/cirosantilli/linux-kernel-module-cheat#supported-hosts

Thank for your work, looking forward to reading more!

cirosantilli, 2020-03-22:
Hi Max,

This is because of: https://github.com/cirosantilli/linux-kernel-module-cheat/issues/75

See workarounds at: https://github.com/cirosantilli/linux-kernel-module-cheat#linux-kernel-module-cheat

mbigras, 2020-03-22:
Ah makes sense, thanks for the links, as mentioned in https://github.com/cirosantilli/linux-kernel-module-cheat/issues/75 I can view the full READMe at:

https://github.com/cirosantilli/linux-kernel-module-cheat/blob/master/README.adoc

closed 2020-03-22 by mbigras