cirosantilli/linux-kernel-module-cheat #120: Simple kernel boot parameter question

tuinannan, 2020-03-25:
hi ciro,

how do variables get passed from the boot parameter to init scripts?

i unfortunately need to modify an image that is not based on this repo.

i'm curious about how, for example, lkmc_eval_base64 gets passed to the S98 init script.

is this because this is running is the same shell context?

also, is it necessary for variables to be passed after the dash '-' , e.g. in qemu boot parameter append:
```
-append 'root= ... - s98 params'
```

thanks

cirosantilli, 2020-03-25:
Hi Yanan,

I think this is well explained in this section and the ones that follow it: https://cirosantilli.com/linux-kernel-module-cheat/#init-environment

I've just reviewed it and fixed a minor error in the tutorial at c5aeb1791c00ce7a41c79e47658c681e676ee65e

If something is not clear after reading that and playing with `init_env_poweroff.out`, let me know.

tuinannan, 2020-03-26:
谢谢

closed 2020-03-26 by tuinannan