cirosantilli/linux-kernel-module-cheat #121: No module named 'china_dictatorship'

LuminousXLB, 2020-03-30:
The module was imported [here](https://github.com/cirosantilli/linux-kernel-module-cheat/blob/master/common.py#L31) and only used [here](https://github.com/cirosantilli/linux-kernel-module-cheat/blob/master/common.py#L1361).

I don't know what it is, nor can I install it with pip. But since it is not vital, it might be better to move the `import` thing under `if env['china']:`

cirosantilli, 2020-03-30:
Oh no! My attempt to have some fun backfired! 

0ee5547365a517ab7fec08513fb548dd8bb5daff adds that by default on `./build --download-dependencies` and for now you can work around it with:

```
pip3 install -r requirements.txt
```

closed 2020-03-30 by cirosantilli

cirosantilli, 2020-03-30:
Further made it optional at 112be2fa50e45efc737ad80f2975d0074e433cf7