cirosantilli/linux-kernel-module-cheat #122: Unable to run buildroot for gem5

GeoBK, 2020-04-06:
In reference to the answer here on stackoverflow - https://stackoverflow.com/questions/60942236/is-there-an-updated-disk-image-binary-for-the-x86-architecture-for-running-gem5 - 
Running the command 
./build --download-dependencies gem5-buildroot
and then running 
./run --emulator gem5
the second command fails saying - 
"No such file or directory: '/home/bibin/projects/linux-kernel-module-cheat/out/gem5/default/build/X86/gem5.opt"
Is there some additional command which needs to be run as well before running the gem5 emulator?
I am running on an Ubuntu18.04.4 LTS system

cirosantilli, 2020-04-06:
Hi geo,

Those two should be it.

Can you double check that the `./build` command succeed without errors? When I do a `./build --dry-run` it does show the gem5 build commands (search for `gem5.opt`), so it feels like the build failed for some reason.

yunchih, 2020-04-08:
Gem5 failed to compile:
```
 [SO PARAM] PS2Device -> X86/params/PS2Device.hh
 [     CXX] X86/dev/x86/i8254.cc -> .o
In file included from /cheat/linux-kernel-module-cheat/out.docker/gem5/default/build/X86/base/stats/hdf5.cc:38:
/cheat/linux-kernel-module-cheat/out.docker/gem5/default/build/X86/base/stats/hdf5.hh:41:10: fatal error: H5Cpp.h: No such file or directory
 #include <H5Cpp.h>
          ^~~~~~~~~
compilation terminated
```

can be resolved by manually adding include directory into `SConstruct`:

```
main.Prepend(CPPPATH=Dir('/usr/include/hdf5/serial'))
```

cirosantilli, 2020-04-08:
@yunchih thanks. The main scons attempts to take care of that and build works for me without patching: https://github.com/gem5/gem5/blob/9fc9c67b4242c03f165951775be5cd0812f2a705/SConstruct#L871 but maybe there's a case where it breaks.

GeoBK, 2020-04-13:
I am not sure - but my theory right now is that my system might have run out of memory and therefore maybe it wasn't able to generate the gem5.opt file. I had around 17GB of free memory on that partition when I started the build. I'll try running the build again when I get access to that machine again (I cant physically access it right now with the quarantine and all)

Jrebort, 2020-09-17:
I have same question
`./linux-kernel-module-cheat/out/gem5/default/build/X86` Gem5.opt does not seem to be generated here.
I tried to run SConscript in this directory, but got the following error
```
./SConscript:45: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Traceback (most recent call last):
  File "./SConscript", line 56, in <module>
    from gem5_scons import Transform, warning, error
ModuleNotFoundError: No module named 'gem5_scons'
```

cirosantilli, 2020-09-17:
@Jrebort hi, can you try to `sudo apt install `libhdf5-dev` and rebuild gem5 (`./build-gem5`). I've noticed that this is missing, and I'm adding it to the dependencies.

I also noticed that `./build` does not stop on errors which is super confusing (likely `./build` said the gem5 build failed, but then printed a gazillion other messages from other builds) I'll also try to address that now.

cirosantilli, 2020-09-17:
@Jrebort  I'm going to try that out. The first suspicion is python2 vs python3 issues due to more or less recent migration. What's your OS version/Python version?

Jrebort, 2020-09-17:
> @Jrebort I'm going to try that out. The first suspicion is python2 vs python3 issues due to more or less recent migration. What's your OS version/Python version?

/usr/bin/python -V

cirosantilli, 2020-09-17:
@Jrebort can you open an issue upstream with the raw scons build command, your OS version, Python version, error message: https://gem5.atlassian.net/jira/software/c/projects/GEM5/issues

I have gem5 compilation working on my Ubuntu 20.04 host in which `python` is now python3.

But I wasn't able to get it working on Ubuntu 20.04 Docker in this repo nicely easily now.

I think I needed to install python2 pip and the module `six` in Python2 on my host, and that required pyenv.

But upstream testing does seem to use docker and no Python2: https://askubuntu.com/questions/350475/how-can-i-install-gem5/1275773#1275773

ARGHH this Python 2 vs 3 is such a mess.

cirosantilli, 2020-09-17:
I'm going to close this ticket for now, because the error message could be too many possibilities: it just means gem5 build failed, which there are many possible failures for.

At bbfa782736dfda014dd9b45f26315e0c666443e2 I've also made `./build` stop on failures, which will make it much easier to notice them :-)

So if anyone gets this problem again, please check the more precise error cause in `./build-gem5` and open a separate issue for that so we can tackle it.

closed 2020-09-17 by cirosantilli