cirosantilli/linux-kernel-module-cheat #123: FORCE_UNSAFE_CONFIGURE=1

smalinux, 2020-04-24:
I don't know what i should do to bypass this error
I used this cmd: `sudo ./build --download-dependencies qemu-buildroot`
and i'm using fedora kernel 5.5.17-200.fc31.x86_64
`
checking whether mknod can create fifo without root privileges... configure: error: in /home/smalinux/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-tar-1.29':
configure: error: you should not run configure as root (set FORCE_UNSAFE_CONFIGURE=1 in environment to bypass this check)
See `config.log' for more details
make: *** [package/pkg-generic.mk:231: /home/smalinux/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/host-tar-1.29/.stamp_configured] Error 1
Traceback (most recent call last):
  File "./build", line 659, in <module>
    Main().cli()
  File "/home/smalinux/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/smalinux/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/smalinux/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/smalinux/linux-kernel-module-cheat/common.py", line 1484, in main
    ret = self.timed_main()
  File "./build", line 656, in timed_main
    component.build(self.env['arch'])
  File "./build", line 57, in build
    self.build_callback()
  File "./build", line 449, in f
    lkmc.import_path.import_path_main(component_file)(**args)
  File "/home/smalinux/linux-kernel-module-cheat/common.py", line 719, in __call__
    return super().__call__(**kwargs)
  File "/home/smalinux/linux-kernel-module-cheat/cli_function.py", line 149, in __call__
    return self._do_main(kwargs)
  File "/home/smalinux/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/smalinux/linux-kernel-module-cheat/common.py", line 1484, in main
    ret = self.timed_main()
  File "/home/smalinux/linux-kernel-module-cheat/common.py", line 1981, in timed_main
    return self.build()
  File "/home/smalinux/linux-kernel-module-cheat/build-buildroot", line 157, in build
    cwd=self.env['buildroot_source_dir'],
  File "/home/smalinux/linux-kernel-module-cheat/shell_helpers.py", line 444, in run_cmd
    raise e
Exception: Command exited with status: 2
`

closed 2020-04-24 by smalinux

cirosantilli, 2020-04-24:
I'm not sure what happened, but please don't run `./build` with `sudo` :-)

smalinux, 2020-04-24:
Okey. thank you.  
I used `export FORCE_UNSAFE_CONFIGURE=1` with `chmod -R 777 ./` as error said and it has gone :D