cirosantilli/linux-kernel-module-cheat #124: Specific Linux Kernel build?

dlannan-fmad, 2020-06-19:
Id like to update the kernel to a specific kernel version (5.4.19).
I read the documents, but as I understand it (please correct me) is that only the patched branches are available for the lkmc?
This is ok, if its the case, I'll put together my own 'half assed' qemu kernel build :)
If there is a way (because I love your toolkit btw), please let me know the process.

dlannan-fmad, 2020-06-19:
I think I have figured it out. Its pretty simple (Im assuming because the major version is the same). 
What I did (really not sure if this is right)
- Copy linux (5.4.19) into submodules/linux (backed up the linux folder in case)
- Run ./build_linux
Seems good so far.

closed 2020-06-19 by dlannan-fmad

cirosantilli, 2020-06-19:
Hi David,

That's basically it, see also: https://cirosantilli.com/linux-kernel-module-cheat/#linux-kernel-build-variants

Another thing which would be cool would be to generalize --gem5-worktree https://cirosantilli.com/linux-kernel-module-cheat/#gem5-worktree to all `./build-*` commands, but not done yet.

dlannan-saldo, 2020-06-21:
Thanks very much. Really appreciate the help. Excellent toolkit.