cirosantilli/linux-kernel-module-cheat #129: Failed to build gem5

Jrebort, 2020-09-17:
When I was running the build gem5 command, I got the following error. I tried to solve it but failed. please help me.

scons: Entering directory `/home/yangshuang/workspace/linux-kernel-module-cheat/submodules/gem5/util/m5'
scons: Reading SConscript files ...

scons: warning: Two different environments were specified for target m5_mmap.o,
	but they appear to have the same action: $CC -o $TARGET -c $CFLAGS $CCFLAGS $_CCCOMCOM $SOURCES
File "/home/yangshuang/workspace/linux-kernel-module-cheat/submodules/gem5/util/m5/build/aarch64/SConscript", line 48, in <module>
scons: done reading SConscript files.
scons: Building targets ...
scons: *** Do not know how to make File target `build/x86/out/m5' (/home/yangshuang/workspace/linux-kernel-module-cheat/submodules/gem5/util/m5/build/x86/out/m5).  Stop.
scons: building terminated because of errors.
Traceback (most recent call last):
  File "./build", line 659, in <module>
    Main().cli()
  File "/home/yangshuang/workspace/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/yangshuang/workspace/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/yangshuang/workspace/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/yangshuang/workspace/linux-kernel-module-cheat/common.py", line 1518, in main
    ret = self.timed_main()
  File "./build", line 656, in timed_main
    component.build(self.env['arch'])
  File "./build", line 57, in build
    self.build_callback()
  File "./build", line 449, in f
    lkmc.import_path.import_path_main(component_file)(**args)
  File "/home/yangshuang/workspace/linux-kernel-module-cheat/common.py", line 733, in __call__
    return super().__call__(**kwargs)
  File "/home/yangshuang/workspace/linux-kernel-module-cheat/cli_function.py", line 149, in __call__
    return self._do_main(kwargs)
  File "/home/yangshuang/workspace/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/yangshuang/workspace/linux-kernel-module-cheat/common.py", line 1518, in main
    ret = self.timed_main()
  File "/home/yangshuang/workspace/linux-kernel-module-cheat/common.py", line 2011, in timed_main
    return self.build()
  File "/home/yangshuang/workspace/linux-kernel-module-cheat/build-m5", line 32, in build
    self.sh.run_cmd(
  File "/home/yangshuang/workspace/linux-kernel-module-cheat/shell_helpers.py", line 457, in run_cmd
    raise e
Exception: Command exited with status: 2

cirosantilli, 2020-09-17:
Oopsey, forgot to push 76a48621f494add0919be1ca9f0787092f10eb45 which updates gem5, the x86 one was broken previously. Let me know if it works after updating that submodule.

Jrebort, 2020-09-17:
After I pull the updated code, this problem was solved, thank you very much!

cirosantilli, 2020-09-17:
Thanks for the report! Break fast, fix fast :-)

closed 2020-09-17 by cirosantilli