cirosantilli/linux-kernel-module-cheat #135: qemu cloning failure

ashriram, 2020-11-23:
fatal: remote error: upload-pack: not our ref d0f9ab350f315aa3643791b0587d28b38bf3b18c
Fetched in submodule path 'submodules/qemu', but it did not contain d0f9ab350f315aa3643791b0587d28b38bf3b18c. Direct fetching of that commit failed. v3.0 . Ubuntu 18

cirosantilli, 2020-11-23:
ashriram, I had to remove that commit which broke the submodule on older versions, can you just use LKMC master instead?

ashriram, 2020-11-24:
Thanks for the quick response.
Is there a stable version I can use on Ubuntu 18 and supports kernel 4.x
I am trying to use this for a course and the lab machines are all Ubuntu 18.

cirosantilli, 2020-11-24:
Sorry for this mess. I've now created a new version on branch `b3.0` on top of `v3.0` with the submodule fixed. Let me know if that does not work for any reason.

ashriram, 2020-11-24:
Much appreciated; it works.

closed 2020-11-24 by ashriram