cirosantilli/linux-kernel-module-cheat #136: vermagic error

ashriram, 2020-11-24:
on master.

./build -a x --download-dependencies qemu-buildroot

cc1: all warnings being treated as errors
/local-scratch/home/ashriram/build/Linux/lkml/linux-kernel-module-cheat/submodules/linux/scripts/Makefile.build:283: recipe for target '/local-scratch/home/ashriram/build/Linux/lkml/linux-kernel-module-cheat/od
make[3]: *** [/local-scratch/home/ashriram/build/Linux/lkml/linux-kernel-module-cheat/out/kernel_modules/x86_64/kernel_modules/myprintk.o] Error 1
make[3]: *** Waiting for unfinished jobs....
In file included from /local-scratch/home/ashriram/build/Linux/lkml/linux-kernel-module-cheat/out/kernel_modules/x86_64/kernel_modules/vermagic.c:5:
/local-scratch/home/ashriram/build/Linux/lkml/linux-kernel-module-cheat/submodules/linux/include/linux/vermagic.h:6:2: error: #error "This header can be included from kernel/module.c or *.mod.c only"
 #error "This header can be included from kernel/module.c or *.mod.c only"
  ^~~~~
/local-scratch/home/ashriram/build/Linux/lkml/linux-kernel-module-cheat/out/kernel_modules/x86_64/kernel_modules/procfs.c: In function ‘myinit’:
/local-scratch/home/ashriram/build/Linux/lkml/linux-kernel-module-cheat/out/kernel_modules/x86_64/kernel_modules/procfs.c:32:40: error: passing argument 4 of ‘proc_create’ from incompatible pointer type [-Werr]
  proc_create(filename, 0, NULL, &fops);
                                        ^
In file included from /local-scratch/home/ashriram/build/Linux/lkml/linux-kernel-module-cheat/out/kernel_modules/x86_64/kernel_modules/procfs.c:5:
/local-scratch/home/ashriram/build/Linux/lkml/linux-kernel-module-cheat/submodules/linux/include/linux/proc_fs.h:107:24: note: expected ‘const struct proc_ops *’ but argument is of type ‘const struct file_oper’
 struct proc_dir_entry *proc_create(const char *name, umode_t mode, struct proc_dir_entry *parent, const struct proc_ops *proc_ops);
                        ^~~~~~~~~~~

cirosantilli, 2020-11-24:
Ooop, didn't test the modules very well, I thought I had, some kernel APIs broke:

- `dep.c` and `dep2.c`: `debugfs_create_u32` does not return anymore, not sure why: https://unix.stackexchange.com/questions/593983/creating-a-debugfs-file-that-is-used-to-read-write-u32-value/621282#621282
- vermagic.c: https://github.com/cirosantilli/linux/commit/51161bfc66a68d21f13d15a689b3ea7980457790 prevents its usage in v5.8, not sure why. 
- procfs.c: `struct file_operations` moved to a new `struct proc_ops` at: https://github.com/cirosantilli/linux/commit/b567e075139c68c63f1866df23aebe0e86522899

I'll fix those when I get a moment, should be easy.

cirosantilli, 2020-11-24:
Thanks for this report! Let me know if any other problems arise!

closed 2020-11-24 by cirosantilli