cirosantilli/linux-kernel-module-cheat #138: gem5 build error on ubuntu 16.04

hgedek, 2020-11-30:
os: ubuntu 16.04

I'm getting this error message after I enter: 

./build --download-dependencies gem5-buildroot

...
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'libpng12-dev' instead of 'libpng-dev'
E: Unable to locate package lld
E: Unable to locate package python-is-python3
Traceback (most recent call last):
  File "./build", line 679, in <module>
    Main().cli()
  File "/home/ubuntu/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/ubuntu/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/ubuntu/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/ubuntu/linux-kernel-module-cheat/common.py", line 1533, in main
    ret = self.timed_main()
  File "./build", line 567, in timed_main
    self.sh.add_newlines(sorted(apt_get_pkgs))
  File "/home/ubuntu/linux-kernel-module-cheat/shell_helpers.py", line 457, in run_cmd
    raise e
Exception: Command exited with status: 100

hgedek, 2020-11-30:
When I tried to build on WSL2 - Ubuntu 20.04

It's exiting with this log:

Global frequency set at 1000000000000 ticks per second
warn: No dot file generated. Please install pydot to generate the dot file and pdf.
warn: DRAM device capacity (8192 Mbytes) does not match the address range assigned (256 Mbytes)
info: kernel located at: /home/hgedek/linux-kernel-module-cheat/out/linux/default/x86_64/vmlinux
system.pc.com_1.device: Listening for connections on port 3456
fatal: system.pc.south_bridge.io_apic does not have any port named int_requestor
Memory Usage: 899944 KBytes
time 00:00:02

cirosantilli, 2020-11-30:
Hakan,

About 16.04, I haven't tested with it for a long time, so it could have broken. I think it would likely be easy to fix though. If you manage, do share the solution, and feel free to send a patch that generalizes the build to muiltiple hosts.

I haven't tested x86 gem5 boot in a while. Can you check on gem5 `develop` as well? If it fails, let's open a ticket upstream.

cirosantilli, 2020-11-30:
x86 gem5 boot worked for me on gem5 develop 937241101fae2cd0755c43c33bab2537b47596a2 LKMC 50570326d414bf213f279a791e4ae03210bafa4d let me now if you don't manage to reproduce.