cirosantilli/linux-kernel-module-cheat #139: Multithreading in gem5 aarch64 FS using taskset

itamarbon, 2020-12-02:
I am trying to explore multithreading on a gem5 simulator running through LKMC. I want to run different processes on different CPUs, and am trying to bind the tasks to each CPU using taskset. 
I get `-/bin/sh: taskset: not found` when I try to run the command. 
What can I do to enable this utility in for aarch64 FS.
Thanks,
Itamar

cirosantilli, 2020-12-02:
Ah, now I understood the problem taskset is not built in by default, can you rebuild the image with:

```
./build-buildroot \
  --arch aarch64 \
  --config 'BR2_PACKAGE_UTIL_LINUX=y' \
  --config 'BR2_PACKAGE_UTIL_LINUX_SCHEDUTILS=y' \
;
```

This is mentioned at: https://cirosantilli.com/linux-kernel-module-cheat/#gdb-step-debug-multicore-userland but might not be clear from other sections.

itamarbon, 2020-12-03:
thanks, this works