cirosantilli/linux-kernel-module-cheat #141: What's next step?!

smalinux, 2020-12-29:
Hello, Ciro,
I read most of your amazing lkmc repo and I read almost half of the LDD3 book and I did a lot of random [Linux kernel modules](https://github.com/smalinux/lab).
But I stopped for a moment to ask myself what is the next step to be a professional Linux developer.
I know that I still at the beginning of the road, but I lack the vision that guides me after all that.
What are the options available to dive deeper into the kernel?
Writing more modules, or reading in a specific Septystem?
What's your opinion.
I am currently in my last year of university and I am thinking of participating in GSoC to do more and bigger projects.
Thank you.

cirosantilli, 2020-12-29:
Hi Sohaib,

I think you are on the right track.

Things that come to mind are:

- look for a job in the area, especially in a big company (notably hardware companies) that does kernel dev. Maybe it won't be exactly in Linux kernel development to start with, but from inside the company you might get a better opening. But it is not impossible that you can get a job directly in kernel dev
- Even if you are not an expert now, there might be possible to get hired in the area just by showing your interest/possibly experience so far
- find a hardware that you care about that does not have a Linux driver yet and port it. This is unfortunately hard from outside a company, since most devices will come from the company (except reverse engineered ones if your are into that sort of stuff)
- find a non hardware specific project to try and help. There are slightly higher level parts of the kernel in which more experimental patches are being done to see if there are perf/functionality gains. I would have a look to see if anyone from your university has any related research projects, finding such ideas is basically like finding a research project. Maybe something like BPF/some new networking capability/some new filesystem capability.

Good luck and let me know how it goes.

smalinux, 2021-07-12:
Hey @cirosantilli 
> Good luck and let me know how it goes.

just some changelog I think I'm now at a good point to talk about them.
After I talk to you I decided to try to participate in Google Summer of Code.
I messaged many mentors and I got many positive (thanks to this repo as my CV became much better because of it), 
let me share with you some of them:  

[image]

[image]

After that, I talked to the Performance Co-Pilot mentor and he was very supportive of me. so I started directly contributing with PCP and Htop  [Link](https://github.com/htop-dev/htop/pull/669)
nowadays I contribute with Performance Co-Pilot but not under the GSoC banner because PCP org has not been given a spot for GSoC this year I'll apply again next year, and maybe I get a better result

two days ago my first patch to Linux kernel get accepted 🙂 :
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?id=4f118472d4b40142004bc6bc652a651f67bfee94

So I'm now very happy with my contributions 🙂and let me close this issue as I have a better vision now.

thanks so much, Ciro

closed 2021-07-12 by smalinux

cirosantilli, 2021-07-12:
Nice! Really glad things have been moving forward!! :tada: :tada: :tada: :tada: :tada: :tada: :tada: :tada: :tada: :tada: