cirosantilli/linux-kernel-module-cheat #142: can't compile qemu. it says "can't find ninja"

SeannnnK, 2021-01-18:
Hi Ciro,

I met the below problem and I searched the issues which don't include this one. You can take a look below error:

Thanks a lot.

/**********************************************************************************************************************/
+ cd /home/sean/linux-kernel-module-cheat/out/qemu/default/opt && \                                                                                                                                       
  PATH='/usr/lib/ccache:"${PATH}"' \                                                                                                                                                                       
  /home/sean/linux-kernel-module-cheat/submodules/qemu/configure \                                                                                                                                        
  --enable-trace-backends=simple \                                                                                                                                                                         
  --target-list=x86_64-softmmu \                                                                                                                                                                           
  --enable-sdl \                                                                                                                                                                                           
;

ERROR: Cannot find Ninja

Traceback (most recent call last):
  File "./build", line 679, in <module>
    Main().cli()
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs) 
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/sean/linux-kernel-module-cheat/common.py", line 1540, in main
    ret = self.timed_main()
  File "./build", line 674, in timed_main
    ret = component.build(self.env['arch'])
  File "./build", line 55, in build
    return self.build_callback()
  File "./build", line 471, in f
    return lkmc.import_path.import_path_main(component_file)(**args)
  File "/home/sean/linux-kernel-module-cheat/common.py", line 739, in __call__
    return super().__call__(**kwargs)
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 149, in __call__
    return self._do_main(kwargs)
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/sean/linux-kernel-module-cheat/common.py", line 1540, in main
    ret = self.timed_main()
  File "/home/sean/linux-kernel-module-cheat/common.py", line 2059, in timed_main
    return self.build()
  File "/home/sean/linux-kernel-module-cheat/build-qemu", line 38, in build
    self.sh.run_cmd(
  File "/home/sean/linux-kernel-module-cheat/shell_helpers.py", line 457, in run_cmd
    raise e
Exception: Command exited with status: 
 if self.env['configure']:
            self.sh.run_cmd(
                [   
                    os.path.join(self.env['qemu_source_dir'], 'configure'), LF, 
                    '--enable-trace-backends=simple', LF, 
                    '--target-list={}'.format(target_list), LF, 
                    '--enable-sdl', LF, 
                ] + 
                build_type_cmd +
                self.sh.shlex_split(self.env['extra_config_args']),
                extra_paths=[self.env['ccache_dir']],
                cwd=build_dir
            )

/**********************************************************************************************************************/

cirosantilli, 2021-01-18:
QEMU moved to ninja apparently, can you try:

```
sudo apt install ninja-build
```

SeannnnK, 2021-01-18:
Thanks. It fixed. 

You can close this issuse.

> QEMU moved to ninja apparently, can you try:
> 
> ```
> sudo apt install ninja-build
> ```
> 
> Also added at [f350d28](https://github.com/cirosantilli/linux-kernel-module-cheat/commit/f350d28a808dbf0dd374197bfb33e6940da093c9) now.

Thanks. Resolved.

cirosantilli, 2021-01-18:
:100:

closed 2021-01-18 by cirosantilli

vimtomk, 2022-04-28:
It should be noted, on MacOS, you can install this package using `brew install ninja`.