cirosantilli/linux-kernel-module-cheat #143: compileing glibc met the error: "make[3]: *** No rule to make target 'elf/soinit.c', needed by '/home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/glibc-custom/build/elf/soinit.os'.  Stop."

SeannnnK, 2021-01-19:
Hi Ciro,

Sorry for the compiling issue? Could you take a look on the compiling issue on glibc?

Thanks for you help.

**************************************************************************************************************************************************
/usr/bin/make  subdir=elf -C elf ..=../ subdir_lib
make[3]: Entering directory '/home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/glibc-custom/elf'
cd /home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/glibc-custom/build && /home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/host/lib/gcc/x86_64-build`
make[3]: *** No rule to make target 'elf/soinit.c', needed by '/home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/glibc-custom/build/elf/soinit.os'.  Stop.
make[3]: *** Waiting for unfinished jobs....
r - libio/ioputs.os
make[3]: Leaving directory '/home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/glibc-custom/elf'
make[2]: *** [Makefile:259: elf/subdir_lib] Error 2
make[2]: Leaving directory '/home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/glibc-custom'
make[1]: *** [Makefile:9: all] Error 2
make[1]: Leaving directory '/home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/glibc-custom/build'
make: *** [package/pkg-generic.mk:250: /home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/build/glibc-custom/.stamp_built] Error 2
Traceback (most recent call last):
  File "./build-buildroot", line 178, in <module>
    Main().cli()
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/sean/linux-kernel-module-cheat/common.py", line 1540, in main
    ret = self.timed_main()
  File "/home/sean/linux-kernel-module-cheat/common.py", line 2059, in timed_main
    return self.build()
  File "./build-buildroot", line 146, in build
    self.sh.run_cmd(
  File "/home/sean/linux-kernel-module-cheat/shell_helpers.py", line 457, in run_cmd
    raise e
Exception: Command exited with status: 2

cirosantilli, 2021-01-24:
Sean, thanks for the report, to hesitate to create issues here.

But I'm afraid I couldn't reproduce this one, I did a clean:

```
rm -rf out/
./build
```

on f350d28a808dbf0dd374197bfb33e6940da093c9 (I already had the dependencies installed from a previous `./build --download-dependencies` though) and it worked.

Can you double check that your build directory is clean before running the command?

SeannnnK, 2021-01-25:
Santilli, thanks for your reply. I did do the step to clean the build. But I failed to compile the glibc. Let me do it again to reproduce it.I also git pull to update the repo. I went into the glibc submoduls library and run some compile command. I still met this issue. 

In order to resolve this issue I also googled for this one.  I am a newbie on the linux system and kernel.  I found the similar issue on https://stackoverflow.com/questions/9853528/error-while-building-glibc. But I failed to compile too.

Let me git pull, run clean command and build it again. And try to give more info.

SeannnnK, 2021-01-25:
By the way I saw this error when I run the ./build. I though this is related with the persimisson. So I run sudo ./run

************************************************************************************************************************************************

+ cd /home/sean/linux-kernel-module-cheat/submodules/buildroot && \
  FORCE_UNSAFE_CONFIGURE=1 \
  make \
  'LKMC_PARSEC_BENCHMARK_SRCDIR="/home/sean/linux-kernel-module-cheat/submodules/parsec-benchmark"' \
  O=/home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64 \
  V=0 \
  all \
;
/usr/bin/make -j1 O=/home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64 HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
  GEN     /home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/Makefile

You seem to have the current working directory in your
PATH environment variable. This doesn't work.
make[1]: *** [support/dependencies/dependencies.mk:27: dependencies] Error 1
make: *** [Makefile:84: _all] Error 2
Traceback (most recent call last):
  File "./build", line 682, in <module>
    Main().cli()
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/sean/linux-kernel-module-cheat/common.py", line 1540, in main
    ret = self.timed_main()
  File "./build", line 677, in timed_main
    ret = component.build(self.env['arch'])
  File "./build", line 55, in build
    return self.build_callback()
  File "./build", line 474, in f
    return lkmc.import_path.import_path_main(component_file)(**args)
  File "/home/sean/linux-kernel-module-cheat/common.py", line 739, in __call__
    return super().__call__(**kwargs)
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 149, in __call__
    return self._do_main(kwargs)
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/sean/linux-kernel-module-cheat/common.py", line 1540, in main
    ret = self.timed_main()
  File "/home/sean/linux-kernel-module-cheat/common.py", line 2059, in timed_main
    return self.build()
  File "/home/sean/linux-kernel-module-cheat/build-buildroot", line 146, in build
    self.sh.run_cmd(
  File "/home/sean/linux-kernel-module-cheat/shell_helpers.py", line 457, in run_cmd
    raise e