cirosantilli/linux-kernel-module-cheat #144: file not found error: qemu-x86_64

SeannnnK, 2021-01-26:
Santilli, I thought the commnad name "qemu-x86-64" should be changed into "qemu-system-x86_64"

I run the below commnad  ./run --userland userland/c/hello.c and it gived the below errror:

sean@sean-ThinkPad:~/linux-kernel-module-cheat$ ./run --userland userland/c/hello.c
+ /home/sean/linux-kernel-module-cheat/out/qemu/default/opt/x86_64-linux-user/qemu-x86_64 \
  -L /home/sean/linux-kernel-module-cheat/out/buildroot/build/default/x86_64/target \
  -r 5.9.1 \
  -seed 0 \
  -trace enable=load_file,file=/home/sean/linux-kernel-module-cheat/out/run/qemu/x86_64/0/trace.bin \
  -cpu max \
  /home/sean/linux-kernel-module-cheat/out/userland/default/x86_64/c/hello.out \
;
Traceback (most recent call last):
  File "./run", line 999, in <module>
    Main().cli()
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 267, in cli
    exit_status = self.cli_noexit(*args, **kwargs)
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 258, in cli_noexit
    return self._do_main(vars(args))
  File "/home/sean/linux-kernel-module-cheat/cli_function.py", line 152, in _do_main
    return self.main(**self._get_args(kwargs))
  File "/home/sean/linux-kernel-module-cheat/common.py", line 1540, in main
    ret = self.timed_main()
  File "./run", line 942, in timed_main
    exit_status = self.sh.run_cmd(
  File "/home/sean/linux-kernel-module-cheat/shell_helpers.py", line 414, in run_cmd
    with subprocess.Popen(
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/sean/linux-kernel-module-cheat/out/qemu/default/opt/x86_64-linux-user/qemu-x86_64'
sean@sean-ThinkPad:~/linux-kernel-module-cheat$ find . -name qemu-x86_64
sean@sean-ThinkPad:~/linux-kernel-module-cheat$ 
sean@sean-ThinkPad:~/linux-kernel-module-cheat$ 
sean@sean-ThinkPad:~/linux-kernel-module-cheat$ 
sean@sean-ThinkPad:~/linux-kernel-module-cheat$ find . -name qemu-system-x86_64
./out/qemu/default/opt/x86_64-softmmu/qemu-system-x86_64
./out/qemu/default/opt/qemu-system-x86_64
sean@sean-ThinkPad:~/linux-kernel-module-cheat$

cirosantilli, 2021-01-26:
Hi Sean,

QEMU has two separate executables:

- `qemu-x86_64` for userland
- `qemu-system-x86_64` for full system

To build just userland with this repo you can do:

```
./build-qemu --mode userland
```

or to also build other userland stuff as well:

```
./build user-mode-qemu
```

SeannnnK, 2021-01-26:
> Hi Sean,
> 
> QEMU has two separate executables:
> 
> * `qemu-x86_64` for userland
> * `qemu-system-x86_64` for full system
> 
> To build just userland with this repo you can do:
> 
> ```
> ./build-qemu --mode userland
> ```
> 
> or to also build other userland stuff as well:
> 
> ```
> ./build user-mode-qemu
> ```

Thanks for your help.

closed 2021-01-26 by SeannnnK