cirosantilli/linux-kernel-module-cheat #145: gem5 now build with arm64 command.

EE415-20170470, 2021-02-27:
[image]

I have error on running `./build --arch aarch64 --download-dependencies gem5-buildroot command because gem5 now build with arm64 command!`

I also has same error on #143 because I think some environment missed when running only `./build-buildroot -- glibc-reconfigure`

cirosantilli, 2021-02-27:
Can you try:

```
sudo apt install liblua5.4-dev
```

and then check with:

```
./build-m5 -aA
```

? The m5 has an annoying dependency on lua I believe.

EE415-20170470, 2021-02-28:
I think you can just change 

```
        if self.env['arch'] == 'x86_64':
            env['gem5_m5_source_dir_build_arch'] = 'x86'
        else:
            env['gem5_m5_source_dir_build_arch'] = env['arch']
```

to

```
        if self.env['arch'] == 'x86_64':
            env['gem5_m5_source_dir_build_arch'] = 'x86'
        elif env['host'] == 'aarch64':
            env['gem5_m5_source_dir_build_arch'] = 'arm64'
        else:
            env['gem5_m5_source_dir_build_arch'] = env['arch']
```

in common.py.

It solved my error.

cirosantilli, 2021-03-25:
Thanks for the patch!!!

closed 2021-03-25 by cirosantilli